XCSoar/XCSoar #871: Map download on Kobo no working

hjr, 2022-02-12:
XCSoar versions having and not having the problem
-------------------------------------------------
XCSoar 7.23 git#bc64cat1db  - @kobo mini, @kobo glo 
XCSoar 7.23  - @kobo mini, @kobo glo 
XCSoar 7.22  - @kobo mini
XCSoar 7.21  - @kobo mini

Expected behavior
-----------------
Having the devices hooked to Wifi the Filemanager can download and update maps through internet connection.

Actual behavior
---------------
The update results in an error, maps do not get updated.

Steps to reproduce the behavior
-------------------------------
1. Connect to Wifi
2. Start XCS in fly or sim mode
3. Go to the file manager and see the hints "update available" (so a index file could be accessed through the connection)
4. Hit updated all, or add, or download.
5. The map gest "queued", but no progress shown, instead "error" is displayed.

Do you have any idea what may have caused this?
-----------------------------------------------
no

Do you have an idea how to solve the issue?
-------------------------------------------
no, did check with HEAD on Linux, that works all fine.

lordfolken, 2022-02-13:
Do you have an xcsoar.log? Although i'm not sure there is much info in it. 
XCSoarData doesn't contain the file?
Do curl or wget from shell work?

GeekGoneOld, 2022-02-13:
FWIW, on a clean start on my Clara HD, the last entry in xcsoar.log is:

[2022-02-08T07:05:41Z] CURL failed: error setting certificate file: /etc/ssl/certs/ca-certificates.crt

lordfolken, 2022-02-13:
So the certificate authorities are missing from the koboroot.tgz. Hence no certificate validation of https urls, hence no download.

lordfolken, 2022-02-13:
https://wiki.mozilla.org/CA/Included_Certificates is mozilla's cert store. This is provided to be included in 3rd party applications. 
@hjr can you: 
wget https://ccadb-public.secure.force.com/mozilla/IncludedRootsPEMTxt?TrustBitsInclude=Websites and add the output to /etc/ssl/certs/ca-certificates.crt ? And confirm that this fixes it?

lordfolken, 2022-02-14:
i believe this is simply missing. At the beginning kobo couldn' t do ssl at all. Now it has support but the ca file is missing for validation to be successful. 

And what has happend is that the repository moved to ssl for most of the stuff, so the error pops up now more frequently. 

To implement: 
 I would suggest to add the mozilla catrust to tthe root.tgz. 

For that to happen: we need to download it in build/thirdparty.mk

and deploy it to the root in build.kobo.mk

GeekGoneOld, 2022-02-14:
> To implement: I would suggest to add the mozilla catrust to tthe root.tgz.
> 
> For that to happen: we need to download it in build/thirdparty.mk
> 
> and deploy it to the root in build.kobo.mk

Seems to me thirdparty.mk is centered around building the library, not downloading.  Could the cert be downloaded in kobo.mk or is that bad practice?

> 
> _update_: as an alternative i found that curl has a compile time option: ` ` ` --with-ca-fallback Use the built in CA store of the SSL library` ` `

So what would the best way to use this be?  Do we compile a new version of curl in thirdparty.mk and deploy in kobo.mk?  What about wget?  Do we have to do something with it as well or is it not used by XCSoar or is it a wrapper for curl?  Not really sure where to go from here.

fraktal-sb, 2022-02-17:
confirm this bug also an KOBO GLO HD. Wifi in general works on this device only with the oldest kobo firmware which is loaded after a firmware reset. Question is wether it will be necessary to upgrade to the newest firmware after this bug is corrected.

GeekGoneOld, 2022-02-17:
It is not necessary to upgrade to the newest firmware.  After a firmware reset, you  install XCSoar then using telnet you execute the following command:

wget "https://ccadb-public.secure.force.com/mozilla/IncludedRootsPEMTxt?TrustBitsInclude=Websites" -O /etc/ssl/certs/ca-certificates.crt

(above is all one line)

After that, it will work fine until the next Kobo firmware reset (never?)

lordfolken, 2022-05-19:
Thx to @GeekGoneOld we have a fix in #874

closed 2022-05-19 by lordfolken