XCSoar/XCSoar #872: SDL builds showing black terrain

hjr, 2022-02-13:
XCSoar versions having and not having the problem
-------------------------------------------------
HEAD rev. since somewhen past v7.22 tag.

Expected behavior
-----------------
XCS should show the same nice terrain shading as the native build w/o SDL.
[image]

Actual behavior
---------------
Instead it shows only a black background on the map.
[image]

Steps to reproduce the behavior
-------------------------------
Build w/ make ENABLE_SDL=y USE_SDL2=y or ENABLE_SDL=y. Terrain and Topo ON, any color scheme. My Laptop has an integrated intel graphics adapter. GLE=1.4 GLX=1.4, a bit older, but I would expect XCS to render fine.

Do you have any idea what may have caused this?
-----------------------------------------------
Maybe some not supported GL API.

Do you have an idea how to solve the issue?
-------------------------------------------
Maybe it is not really required to build on more recent GL API.

hjr, 2022-02-18:
Thanks Max for digging in here. Just checked a fresh SDL (1) build with the fix and got this. You might reopen this issue?

[image]

MaxKellermann, 2022-02-19:
It's .... white now! Is that an improvement? ;-)
What build options? Paste your EGL/GLES data from your xcsoar.log

hjr, 2022-02-23:
build w/ make ENABLE_SDL=y 

~/projekte/XCSoar$ ./output/UNIX/bin/xcsoar 
Starting XCSoar XCSoar v7.24-Linux~git#812d86055b
GLX config: RGB=8/8/8 alpha=0 depth=24 stencil=8
GL vendor: Intel Open Source Technology Center
GL version: OpenGL ES 2.0 Mesa 21.3.6
GL renderer: Mesa DRI Intel(R) HD Graphics (ILK)
GL extensions: GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_float_linear GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_NV_conditional_render GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_robustness GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_robustness GL_NV_pixel_buffer_object GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_polygon_offset_clamp GL_EXT_texture_border_clamp GL_KHR_blend_equation_advanced GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_EXT_texture_compression_s3tc_srgb GL_EXT_clip_control GL_EXT_color_buffer_half_float GL_EXT_texture_mirror_clamp_to_edge GL_KHR_parallel_shader_compile GL_EXT_shader_framebuffer_fetch_non_coherent GL_EXT_demote_to_helper_invocation GL_EXT_depth_clamp GL_MESA_bgra 
Display dpi=96,96
OpenGL: mda=1 npot=1 stencil=0x8d48
Startup dialog
Loading profiles

fasiebenwurst, 2022-03-07:
Having the same issue on my RPi4 1GB... XCSoar 7.23 
any hints?

lordfolken, 2022-04-03:
Adding the breaking tag, as it seems to affect rpi4s.

closed 2022-07-14 by lordfolken