XCSoar/XCSoar #873: TCP client not working on Android 11

nkronig, 2022-02-17:
When I connect to a TCP server (for Flarm, LXNAV and KTR2) on any Android 10 phone (tested with 5 different), it works perfectly fine. On my new Android 11 Phone, it cant connect for some reason. After multiple tries and some times just letting it sit there for some Minutes it can connect, but cant receive data. I have tried everything in Androids options, but cant find anything, comming close to being the problem.  Maybe there is some new authorization, that has to be requested? I am not really experienced with developing on Android, but it could be something along those lines.

lordfolken, 2022-02-17:
Do you have a firewall / vpn software installed on your android device? Potentially you need to allow xcsoar to create tcp connections to lan/wan/wifi.

nkronig, 2022-02-21:
I am able to connect to the TCP Client with Android 11, as well as Android 9 (on two Emulated devices). My OnePlus Phone still isnt able to connect. I also tried my brothers Phone( allso OnePlus with Android 11) and it cant connect ether. All other Phones (with Android 9) can. That leads me to conclude, that the Problem could be some OnePlus Specific thing. I have installed a Firewall app from the PlayStore and speciffically granted XCSoar every acess in there. For now this mild Workaround seems to work fine, but it would be Interesting to know, why Oxygen OS seems to be blocking XCSoar in the first place and how to stop it doing that.

lordfolken, 2022-04-26:
As this isn't really an xcsoar issue, i'm closing this. If you still could update this issue in case you have a solution, please update this issue for others to find.

closed 2022-04-26 by lordfolken