XCSoar/XCSoar #875: Error compiling an XCSoar development VM using Vagrant - within Windows

ArgoHG, 2022-02-21:
XCSoar versions having and not having the problem:
Have only tried latest instructions/source as of 21 February 2022 (XCSoar ver 7.23)
-------------------------------------------------

Expected behavior:
For the the command "vagrant up" to successfully build a VirtualBox VM.
-----------------

Actual behavior: VM build starts but then halts with Vagrant error:
XCSoarDevEnv: fatal: repository '/xcsoar-host-src' does not exist

[XCSoar_Vagrant_VM_ScreenOutput_21Feb22.txt](https://github.com/XCSoar/XCSoar/files/8106404/XCSoar_Vagrant_VM_ScreenOutput_21Feb22.txt)
---------------

Steps to reproduce the behavior:
Run Vagrant (for Windows) as per the instructions: https://xcsoar.readthedocs.io/en/latest/build.html - Section: "Using a build VM with Vagrant."
-------------------------------

Do you have any idea what may have caused this?
Perhaps the Windows Host causes the Vagrant script to try to adopt an unsuitable VM directory structure?

I suppose the real question is: Has anyone successfully built a XCSoar development VM using Vagrant from within Windows?
-----------------------------------------------

-------------------------------------------

lordfolken, 2022-02-21:
There is a lot of issues here with trying to do this on windows. Starting with the git checkout that doesn't have the correct line endings, paths etc.

ArgoHG, 2022-02-21:
Many thanks. Clearly I was barking up the wrong o/s tree.
I'll dig out my Linux box and have another crack at it.

lordfolken, 2022-02-22:
Well it shpuld be possible. So it is a valid bug.

ArgoHG, 2022-02-23:
OK, thank you.

lordfolken, 2022-02-23:
As suspected the shell script that is being executed, is with CRLF line endings, and hence cannot be interpreted by the shell inside the vagrant box.

You need to checkout the source code with LF line endings:
First set git to checkout without crlf. NOTE: this is a global option and will affect all git repos!

```bash
git config --global core.autocrlf false
```
Then checkout the source:
```bash
git clone https://github.com/XCSoar/XCSoar.git
```
and run the vagrant up:
```bash
cd ide/vagrant
vagrant up
```
Profit!

(A documentation update would be appreciated.)

lordfolken, 2022-02-23:
If you want to test xcsoar, and run the box interactivly, you need a few things:

vagrant ssh into the box then:
```bash
sudo -i
apt-get install i3 lightdm vim arandr -y
systemctl enable lightdm
systemctl start lightdm
passwd vagrant 
(enter a password x2)
```
then you can login on the graphical console on your VM.
Hit: alt+enter for a terminal 

To change the resolution inside the vm: start a terminal, then start arandr. Choose your desired resolution and hit 'apply'.

ArgoHG, 2022-02-25:
> As suspected, the shell script that is being executed, is with CRLF line endings, and hence cannot be interpreted by the shell inside the vagrant box.
> 
> You need to checkout the source code with LF line endings: First set git to checkout without crlf.
> 
> **NOTE**: this is a global option and will affect all git repos!
> 
> ```shell
> git config --global core.autocrlf false
> ```
> 
> Then checkout the source:
> 
> ```shell
> git clone https://github.com/XCSoar/XCSoar.git
> ```
> 
> and run the vagrant up:
> 
> ```shell
> cd ide/vagrant
> vagrant up
> vagrant ssh
> ```
> 
> Profit!
> 
> (A documentation update would be appreciated)

Thank you.

For my understanding I carried out a XCSoar Android build successfully, via a Linux Host, before coming back to this.

Have now returned to a Windows Host | Vagrant dev VM and followed your instructions. Checking out the source code with LF line endings worked. The source was checked out and then used for building a development Vagrant VM.

Also confirmed the resultant /xcsoar-host-src folder in the VM, is shared with the  /XCSoar folder in the Windows Host - it is.

At this stage however carrying out a make TARGET=ANDROIDAARCH64 within the dev VM fails with building C object errors. Configuring and Generating did complete successfully prior to these building errors. 

Examples of two building errors: 

[5/58] Building C object src/lib/CMakeFiles/c-ares.dir/ares__get_hostent.c.o
FAILED: src/lib/CMakeFiles/c-ares.dir/ares__get_hostent.c.o

[6/58] Building C object src/lib/CMakeFiles/c-ares.dir/ares__parse_into_addrinfo.c.o
FAILED: src/lib/CMakeFiles/c-ares.dir/ares__parse_into_addrinfo.c.o

Copy of build output at first error:

-- Configuring done
-- Generating done
-- Build files have been written to: /home/vagrant/xcsoar-src/output/ANDROID/arm64-v8a/lib/build/c-ares-1.17.1
[1/58] Building C object src/lib/CMakeFiles/c-ares.dir/ares__read_line.c.o
FAILED: src/lib/CMakeFiles/c-ares.dir/ares__read_line.c.o
/home/vagrant/opt/android-ndk-r23b/toolchains/llvm/prebuilt/linux-x86_64/bin/clang --target=aarch64-linux-android -DCARES_STATICLIB -DHAVE_CONFIG_H=1 -D_GNU_SOURCE -D_POSIX_C_SOURCE=199309L -D_XOPEN_SOURCE=600 -I. -I/home/vagrant/xcsoar-src/output/src/c-ares-1.17.1 -I/home/vagrant/xcsoar-src/output/src/c-ares-1.17.1/include -I/home/vagrant/xcsoar-src/output/src/c-ares-1.17.1/src/lib -Os -g -ffunction-sections -fdata-sections -fvisibility=hidden -fpic -funwind-tables -target aarch64-linux-android21 -isystem /home/vagrant/xcsoar-src/output/ANDROID/arm64-v8a/lib/aarch64-linux-android/include -DNDEBUG -I./output/ANDROID/include -DHAVE_POSIX -DHAVE_VASPRINTF -DANDROID -isystem ./output/ANDROID/arm64-v8a/lib/aarch64-linux-android/include -DEYE_CANDY -DBIONIC -DLIBCPP_NO_IOSTREAM  -O3 -DNDEBUG -fPIC -MD -MT src/lib/CMakeFiles/c-ares.dir/ares__read_line.c.o -MF src/lib/CMakeFiles/c-ares.dir/ares__read_line.c.o.d -o src/lib/CMakeFiles/c-ares.dir/ares__read_line.c.o -c /home/vagrant/xcsoar-src/output/src/c-ares-1.17.1/src/lib/ares__read_line.c
In file included from /home/vagrant/xcsoar-src/output/src/c-ares-1.17.1/src/lib/ares__read_line.c:17:
In file included from /home/vagrant/xcsoar-src/output/src/c-ares-1.17.1/src/lib/ares_setup.h:214:
/home/vagrant/xcsoar-src/output/src/c-ares-1.17.1/src/lib/setup_once.h:178:3: error: unknown type name 'Error'
  Error Missing_definition_of_macro_sread
  ^
/home/vagrant/xcsoar-src/output/src/c-ares-1.17.1/src/lib/setup_once.h:178:42: error: expected ';' after top level declarator
  Error Missing_definition_of_macro_sread

...errors repeat for other modules.

Any ideas?

(Point taken about documentation!)

kobedegeest, 2023-01-27:
Thanks a lot was struggling as well with this had same issue as #756. 
Documentation could definatly be improved. 

I think instead of the current:
-Compiling XCSoar
-Setting up development environment 

Which is a bit a weird order cuz you have to set up the environment first. Better would be:
-Getting started
       -Linux users
       -windows users
Where a step by step process can be followed from installing the needed to building XCSoar the first time. Now it feels a bit chaotic (at least for me). 

Making this first step easier to follow for the unexperienced might increase the number of contributors. 
But either way the steps given here are defenitly needed.

lordfolken, 2023-01-28:
Be aware that almost nobody is developing on windows. The docs are cobbled together over the years. If you want to improve things, submit a pr. 
(closing this, as the original problem is a platform issue.)

closed 2023-01-28 by lordfolken