XCSoar/XCSoar #876: Cannot declare task to IMI Erixx logger via USB

kicune, 2022-02-21:
XCSoar versions having and not having the problem
-------------------------------------------------
XCSoar v7.20 on Android (Playstore install)
XCSoar v7.24 on Android (master branch, 812d860)

Which XCSoar version is the last one you know that did not show the problem?
The feature works on XCSoar 7.0_preview15-Android (37a32f2cd3)

Expected behavior
-----------------
The task is successfully declared.

Actual behavior
---------------
After a short while, a window pops up with the message: "Error occured. Task NOT declared!"

This affect only flight decalaration; NMEA data and flight download still work with all app versions.

Steps to reproduce the behavior
-------------------------------
- set  baud rate on IMI Erixx and Android phone (tested with 9600 and 57600 baud - same behavior)
- connect IMI Erixx logger to Android phone (tested with Sony G8441, Android 9) via USB cable and UART converter
- UART converters used: IOIO and USB Serial, both show the same behavior w/ 7.20 and 7.24; success scenario tested only with IOIO, as USB Serial was not available till 7.9
- verify that XCSoar is receiving Baro and GPS data from the logger
- verify that flight download works
- navigate to Manage / Declare, confirm declaration to IMI ERIXX

Do you have any idea what may have caused this?
-----------------------------------------------
Unfortunately, no. Attached please see the communication log files for both failed (with 7.24) and successful (with 7.0) declarations, but my guesses would have a very low added value, I am afraid.

Do you have an idea how to solve the issue?
-------------------------------------------
I am trying to pinpoint the exact version when the feature broke down, but I am having difficulties when compiling older versions on both Docker and Vagrant (atm I am attempting to complile version 7.9, but I am stuck with some Android NDK library version changes that break the compilation for me). I will add more information when/if I resolve these problems.
I can arrange a device lend for testing or participate on testing myself, if it would help.

Many thanks for your effort. If there's anything else I should do or add, please let me know.

kicune, 2022-02-21:
Problem was probably introduced with v7.19 - I compiled both 7.18 and 7.19, and with 7.18 flight declaration works. Flight download with 7.18 is a coin toss - sometimes it works, sometimes it does not, even for the same flight.

With 7.19 the behavior is the same as with the current head: flight download works all the time, but the declaration always fail.
Here are the logs that capture the failed declaration with 7.19 & success with 7.18. Probably not much new there, but just in case.
[xcsoar7.19-declFail.log](https://github.com/XCSoar/XCSoar/files/8110076/xcsoar7.19-declFail.log)
[xcsoar7.18-declOk.log](https://github.com/XCSoar/XCSoar/files/8110077/xcsoar7.18-declOk.log)

lordfolken, 2022-05-02:
@MaxKellermann do you have such a device?

kicune, 2022-08-16:
I traced back the problem to the line 
`return SendRet(port, env, MSG_DECLARATION, &imiDecl, sizeof(imiDecl), MSG_ACK_SUCCESS, 2000, -1) != nullptr;`
The parameter "2000" was probably meant as a value for timeout, but it is in a fact in the position of the parameter `retPayloadSize`. Because of this the `SendRet` never returns.

I created a new pull request #987 that addresses this issue.

closed 2022-08-18 by MaxKellermann