XCSoar/XCSoar #877: Full refresh on kobo screen at set time intervals

Vantuilmv, 2022-02-25:
XCSoar version
--------------

7.23

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

It would be interesting to have a field to inform the interval in seconds for kobo to do a full refresh on the screen, in order to clean the screen that is getting very blurry/ghost  after a long period of flight, for example every 30 seconds a complete refresh leaving everything clean. 

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

I believe it would be very useful to all pilots who make long flights and also who use terrain visualization.

SoaringMLJ, 2022-06-04:
This would also be useful for those of us using vanilla XCSoar on devices with similar screens like the Hisense A5/A7. I have no idea if it would be possible though.

lordfolken, 2026-10-09:
Closing as not planned; periodic e-ink refresh is covered by the display-traits work in #2738. Please reopen with details if still needed.

closed 2026-10-09 by lordfolken