XCSoar/XCSoar #878: Water color customization or color themes

erikhals, 2022-03-01:
XCSoar version
--------------
7.20

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Ability to change water color. 

Brute force solution:
Desaturating and darkening the color of the water in src/Terrain/RasterRenderer.cpp in the RasterRenderer::PrepareColorTable function which is now RawColor(85, 160, 255), a highly saturated blue. The new color needs to be checked for contrast with other map elements. 

Better solution (slightly more work):
Allowing the user to customize the water color in the terrain menu. (create a variable to be imported in RasterRenderer.cpp)

Best solution (a bit more work):
A theme-based approach, similar to terrain colors (which per now do not affect water color, gui or info boxes), but where you can select your preferred color theme from a default selection, or import one:
-classic
-bright color
-dark color
-bright greyscale
-dark greyscale
-high contrast
-e-ink
-OLED
-stamina
which will set the colors of text boxes, terrain colors, airspaces, symbols and lines to a well balanced and carefully selected color palette, optimized for the different screen types in terms of visibility and battery life. 
These colors could then be customized by the user in one color menu, export and import would allow themes to be sharable. 

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
I would like to change the color of the water, which in my opinion is too saturated, too similar in luminance to the terrain and draws attention away from more important elements on the screen. This breaks the design principles, where saturated colors should be reserved for important things. 

It also probably draws more battery on OLEDs than if the color was darker and less saturated. 

For the user coming to the app for the first time, setting up the screen for the best performance on their device can be a bit daunting, as different color schemes are under different menu tabs. There is no way to change water color. 

For developers, importing themes and color variables from one global config file would make more sense than having them hard-coded, as the water color is now.

lordfolken, 2022-03-03:
What you are here referring to is actually the nodata bit. So these are the pixels where the raster has no pixel value (height). Its mostly used for ocean. 
There is more color coding in the map files:for topography: [man](https://mapgen-data.sigkill.ch/manifest) also including water bodies.

Quite a bit of central definition of colors has been done in src/look/ potentially the raster renderer needs a file there.

lordfolken, 2023-02-08:
@erikhals can you come up with a proposal to a good consistent theme?