XCSoar/XCSoar #881: FLARMNet file is parsed incorrectly

lordfolken, 2022-03-03:
XCSoar versions having and not having the problem
-------------------------------------------------
master
7.24

Expected behavior
-----------------
Be shown the following fields in the FlarmTrafficDetails dialog:
- PilotName
- RadioFrequency
- PlaneType

Actual behavior
---------------
The Pilot Name is not displayed with a current flarmnet file. 
The Airport field contains the Registration
[image]

Steps to reproduce the behavior
-------------------------------
* start xcsoar
* download flarmnet data.fln
* go to info->info->traffic list
* search fir i8 or TH (D-44449)

One can also run the FlarmNetTest and see that it fails when run with a current data file. 
```bash
make output/UNIX/bin/TestFlarmNet
wget https://www.flarmnet.org/static/files/wfn/data.fln -O test/flarmnet/data.fln
./output/UNIX/bin/TestFlarmNet
1..15
not ok 1 - count == 6
#     Failed test (test/src/TestFlarmNet.cpp:main() at line 37)
ok 2 - record != NULL
ok 3 - StringIsEqual(record->id, _T("DDA85C"))
not ok 4 - StringIsEqual(record->pilot, _T("Tobias Bieniek"))
#     Failed test (test/src/TestFlarmNet.cpp:main() at line 45)
not ok 5 - StringIsEqual(record->airfield, _T("AACHEN"))
#     Failed test (test/src/TestFlarmNet.cpp:main() at line 46)
ok 6 - StringIsEqual(record->plane_type, _T("Hornet"))
ok 7 - StringIsEqual(record->registration, _T("D-4449"))
ok 8 - StringIsEqual(record->callsign, _T("TH"))
not ok 9 - StringIsEqual(record->frequency, _T("130.625"))
#     Failed test (test/src/TestFlarmNet.cpp:main() at line 50)
not ok 10 - db.FindRecordsByCallSign(_T("TH"), array, 3) == 2
#     Failed test (test/src/TestFlarmNet.cpp:main() at line 53)
not ok 11 - found4449
#     Failed test (test/src/TestFlarmNet.cpp:main() at line 63)
not ok 12 - found5799
#     Failed test (test/src/TestFlarmNet.cpp:main() at line 64)
Segmentation fault
```

Do you have any idea what may have caused this?
-----------------------------------------------
Potentially the export of the flarmnet file has changed. 
The record entry mask on flarmnet.org does only allow the following entries:

radio id | 
Radio id type | ICAO
Pilot name | 
Aircraft type | 
Registration |
Competition id |
Radio frequency | 
Region | 
Livetracking | Active
Last modification | March 2, 2022, 11:49 p.m.

I suspect that region is just for dividing the file, as there are some devices with limited capacity, and flarmnet offeres regional files for those. Its also a only a dropdown, and does not seem to correspond to the "airfield" type record in XCSoar's flarmnet parser. 

Do you have an idea how to solve the issue?
-------------------------------------------
contact support support@flarm.com and ask for the new standard.

lordfolken, 2022-03-03:
@Turbo87 any ideas?

Turbo87, 2022-03-03:
yeah, the flarmnet people stopped publishing these fields due to privacy concerns, which IMHO is stupid because this is voluntarily provided information 🤷‍♂️

Turbo87, 2022-03-03:
I would however recommend to keep those fields supported in xcsoar because there are third party flarmnet files that do fill these fields from other sources like weglide.

what we should do however is check the airport field against the registration and throw the value away if it is the same.

Turbo87, 2022-03-03:
https://github.com/Turbo87/united-flarmnet can produce such a file. I've not cleaned it up properly yet though.

MaxKellermann, 2023-12-21:
So I figure this problem has nothing to do with XCSoar, but is just "shit in shit out"?

closed 2023-12-21 by MaxKellermann