XCSoar/XCSoar #883: XC Soar stops working on Android

RossDouglasMc, 2022-03-07:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2022-03-07:
Do you have a screenshot of the error messgage?

RossDouglasMc, 2022-03-07:
[image]

RossDouglasMc, 2022-03-07:
Just put 7.23 back on, started XC Soar and then minimised the window. When trying to maximise again I got the message "unfortunatley, XC Soar has stopped" (Screenshot above)

MaxKellermann, 2022-03-07:
For further analysis, we need a logcat. This probably shows crash details, and without it, we can't help.

RossDouglasMc, 2022-03-07:
How do I do that?

RossDouglasMc, 2022-03-07:
Just googled it. All too hard. I'll stick with V7.22. Thanks for your time, please close the ticket. 
Best regards
Ross

MaxKellermann, 2022-03-07:
OK, bye, please don't ask for help again.

closed 2022-03-07 by MaxKellermann

RossDouglasMc, 2022-03-07:
Sorry, I didn't mean to be rude at all. My phone doesn't talk to my computer so can't do what they were telling me on Google/Youtube.
I think you guys are great. You have helped with a number of anomolies in the past with exceptional service and I think your product is fabulous. Please don't take offence, I didn't mean that at all.

Best regards
Ross

lordfolken, 2022-03-07:
There are apps that run on your phone that can view logcat locally. 
For example: https://play.google.com/store/apps/details?id=com.tananaev.logcat&hl=en_US&gl=US

RossDouglasMc, 2022-03-07:
Hi Philipp, Just put v7.23 back on, minimised the window then tried to maximise again and got the message  :  This happens every time. CheersRoss From: Philipp WollschlegelSent: Monday, 7 March 2022 3:09 PMTo: XCSoar/XCSoarCc: RossDouglasMc; AuthorSubject: Re: [XCSoar/XCSoar] XC Soar stops working on Android (Issue #883) Do you have a screenshot of the error messgage?

RossDouglasMc, 2022-03-08:
Hi, 

Thank you very much for the link above. The problem I have is that my phone won't connect via USB to a computer. I have spent many hours, on several occassions in the past, trying to troubleshoot this. 

So, after you kindly supplied the link last night, I tried again. I downloaded the app to the phone. I read the instructions. I downloaded Android Studio SDK to my computer. I downloaded more Samsung USB drivers to my computer. I checked that the USB debug was still enabled. I tried every combination of USB protocol on the phone. I tried several USB apps from Play Store on the phone. I tried 2 computers and 3 data cables, one of which is the original supplied with the phone. On every occasion plugging into the computer gets the message "USB device not recognised". Also my car which has Android Play will not recognise this phone. I have in the past tried to resolve this, following a heap of different solutions provided on Google, and have never been able to find anything that worked. I then tried Googling to see if I could find a way to get the Logcat without a USB connection to enter the code. I couldn't find a solution, with several of the forums saying there is no way. I then tired to replicate the problem using the XC Soar I have on my computer for use with Condor. Being a different platform I could not replicate my problem. At 2am, at the end of my limitted ability, in frustration I gave up.

My comment above asking you to cancel the issue must have sounded curt. I certainly didn't mean it that way. I had realised that possibly the problem was my aging phone, and from the information I had and my previous experience, I would not be able to get the information you needed to troubleshoot. I didn't want to further bother you and waste your time on something I was unable to help with. I should have explained that and I do apologise for the wording.

As previoulsy conveyed, I love XC Soar and think it is the best program/app I have come across, bar none. It is my only 'tech' in my aging Jantar. 3 weeks ago and friend and I took a club DG1000 for a cross country. It is fitted with the LX S100, but we both had our phones on the dash running XC Soar because it is just so good. I am hugely appreciative of the support you have provided with the minor issues I have had. Sorry I can't help with this one, and thanks for your efforts.

Best Regards
Ross

P.S. Yes I realise the solution is a new phone :-)