XCSoar/XCSoar #884: Feature Request Altitude Alarms

micheal813, 2022-03-08:
Could a feature be added to have custom altitude alarms? For example, maybe I would like an alarm to go off once at or below a certain altitude to remind me to make sure the landing gear is extended. Maybe be able to create multiple custom alarms?

Thanks for the great app. I use it everytime I fly gliders.

lordfolken, 2022-03-08:
Stuff like this should be possible with the builtin [lua](https://github.com/XCSoar/XCSoar/blob/master/doc/lua.rst) engine.
Maybe you can code a few examples, so we can enhance the docs?

micheal813, 2022-03-08:
> Stuff like this should be possible with the builtin [lua](https://github.com/XCSoar/XCSoar/blob/master/doc/lua.rst) engine. Maybe you can code a few examples, so we can enhance the docs?

Unfortunately I don't know enough about coding to do it myself. Thought about learning how, but not sure if I have the time right now.

lordfolken, 2022-03-08:
This is a community project. If you don't have the time, no one else is there to implement.

lordfolken, 2026-10-22:
Reopening. A Lua script can remind at a height, and that remains fine for an extra custom altitude. An oxygen height, and a gear reminder on the way down, are part of safe gliding and should be settings in XCSoar rather than a script the pilot has to write.