XCSoar/XCSoar #885: Developer Manual is not up-to-date Lua 5.4 and Ubuntu 21 are required

volkmarnissen, 2022-03-08:
XCSoar versions having and not having the problem
-------------------------------------------------
master branch:
The developer Manual doesn't tell anything about  which Ubuntu version to use
it refers to Lua 5.2 as a depenency. This won't work.
Another dependency is G++10.

I used developer Manual 7.0 to build XCSoar from github

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------
Try an older Ubuntu. You will fail because G++10 is required.
the other messages will follow during build process.

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------
Isn't it better to list the known depenencies in the github start page(https://github.com/XCSoar/XCSoar)?
This is easier to maintain.

lordfolken, 2022-03-08:
https://github.com/XCSoar/XCSoar/blob/master/doc/build.rst have a look here. Its sort of the plan to move the Dev manual out of latex. Help wanted.

lordfolken, 2022-03-08:
Also the link in the readme already points to that page.

volkmarnissen, 2022-03-08:
OK. I'll set up my environment for contribution soon. Then I'll add the hint to use ubuntu 21.
Lua 5.4 is already up to date in build.rst.
Let's leave the issue open until I add the ubuntu reference to the build.rst file.

Thank you,

Volkmar

GeekGoneOld, 2022-03-09:
> Then I'll add the hint to use ubuntu 21.

I think Max has been [recommending Debian 11](https://forum.xcsoar.org/viewtopic.php?f=7&t=3954&hilit=debian+bullseye#p22523)

lordfolken, 2022-03-09:
Yes Debian the baseline. We also offer a vagrant box and a docker container with complete toolchains for almost all targets. See: https://github.com/XCSoar/XCSoar/tree/master/ide

MaxKellermann, 2022-05-03:
The compiler dependency is documented, and there is no information about the alleged Lua build problem. This bug report is bogus.