XCSoar/XCSoar #888: Kobo Clara HD does not show built in serial port

GeekGoneOld, 2022-03-12:
On Clara HD the built in serial port (ttymxc0 is not shown in the list for Device port

XCSoar versions having and not having the problem
-------------------------------------------------

Shows up in 7.23.  Clara HD was not supported in previous versions

Expected behavior
-----------------

The port (ttymxc0) should show up in the port list

Actual behavior
---------------

ttymxc0 does not show up in the port list though it clearly shows up in /dev (as seen through wifi/telnet)

Steps to reproduce the behavior
-------------------------------

On Clara HD:
1. Start XCSoar in sim or fly
2. Go to Config | Devices
3. Choose Edit (for any of the six devices
4. Click on the Port
5. Scroll through the list and you will see that ttyxmc0 is missing

Do you have any idea what may have caused this?
-----------------------------------------------

Not yet...  Probably enumeration and filtering of the dev list

Do you have an idea how to solve the issue?
-------------------------------------------

Not yet...

lordfolken, 2022-03-12:
potential duplicate of #879

GeekGoneOld, 2022-03-12:
This might be same as #879.  Presence of port (as seen with Telnet) is key.  Until we know, I will post this here as well:
I noticed that if I changed src/Device/Port/TTYEnumerator.cpp CheckTTYName to filter 'n" to 'z" instead of 'p' to 'w' it works. I have NOT submitted a PR because this isn't really getting to the root of the problem and affects other target platforms.
`@@ -40,7 +40,7 @@ CheckTTYName(const char *name) noexcept

 /* ignore virtual internal ports on Mac OS X (and probably other
    BSDs) */

    if (*t >= 'p' && *t <= 'w')

    if (*t >= 'n' && *t <= 'z')
    return false;
    `

GeekGoneOld, 2022-03-13:
Looks like the Port list of a Device is limited to 128 entries as a 16 bit word is used for each entry where the top 8 bits is the port type and the bottom 8 bits is the index.  Refer to src/Dialogs/Device/PortDataField.cpp.  That rules out increasing the number of entries without major surgery.  That means we have to filter out more tty* entries (as I did for a test).

I need someone experienced in all platforms to decide what we can filter out.  The actual PR is trivial and I'm happy to do the PR and test on a couple of Kobo models but I'm not really equipped to test other platforms.

lordfolken, 2022-03-13:
Just to clarify: The /dev/ttyACM0 character device file is present, but not showing up in XCSoar's Port list?

GeekGoneOld, 2022-03-14:
> Just to clarify: The /dev/ttyACM0 character device file is present, but not showing up in XCSoar's Port list?

That is almost correct.  /dev/ttyACM0 is for issue #879 which MAY be different.  For this issue, I am dealing exclusively with /dev/ttymxc0 and, yes, the character device file is present but it does not show up in XCSoar's Port list.  Specifically, I have determined that the list (of ports) is limited to 128 entries.  If there are more, it drops some from the list.  In particular, on my device the list drops ttya0-3 and ttymxc0.  If I filter out some from being included in the list (, I get ttya0-3 and ttymxc0 in the list.

The filtering is done in src/Device/Port/TTYEnumerator.cpp | CheckTTYName()

It seems to me there are a huge number of tty[a-z][hex digit] entries in the list and I can't quite see their purpose for Kobo (can't speak for Unix, linux or Windows).

cmoetzing, 2022-03-17:
PR #891 fixes the issue for me. Elements above 128 entries were just not added anymore.

samhamill, 2022-04-11:
I’m having the same issue. How can we resolve this?

samhamill, 2022-04-11:
> The correction for this has already been submitted and is awaiting review/approval. Please be patient.
>
> In the meantime, you can select any tty device as Port and create your device. Then go into the appropriate .prf file (usually XCSoarData/default.prf) and edit the port to the correct name (e.g. ttymxc0) then save.

Cheers. I’m quite new to programming but have a basic understanding. What application can be used to Edit the PRF file? Is it pretty straight forward to find the where to edit?

Cheers

GeekGoneOld, 2022-04-12:
>Anyone come across this issue?
> […](#)
Glad to hear you got the workaround to get you through. I’d be very happy to help you with your current problem but the thread is not the best place. Please start a thread on forum.XCSoar.org for the baro issue and we will all be happy to help you out.

GeekGoneOld, 2022-07-28:
This was closed due to a PR but the PR was reversed two weeks later so this should be reopened.

Jeripimo, 2022-07-28:
In addition: Port ttymxc0 not showing in list kobo aura edition 2.
Was showing in earlier versions 6. 
GPS at that port now in latest version 7.23 just says: Port Diasabled  but GPS still working

GeekGoneOld, 2022-08-15:
@MaxKellermann I am not sure why there are so many pseudo ports for the Clara HD.  Could we, perhaps, reduce the number of pseudo ports that are created at startup?

cmoetzing, 2023-05-23:
Since this issue is still open I would like to discuss my initial solution of setting a higher list limit.
Currently only 128 entries are allowed. Therefore ttymxc0 does not show up on my device.
Allowing more entries would fix the problem of having lots of entries until we find a way to filter them correctly.
Since this is only a config dialog that will be accessed only once every now and then this should not be a resource issue if we allow e.g. 256 entries. Right?

GeekGoneOld, 2023-05-30:
I think it would be wise to propose multiple solutions for comment. I propose the following:

1. Remove ports that are not relevant (most tty ports) and leave only port which have a real purpose (includes BT ports)

I have no idea why we have so many tty ports so I have no idea how to reduce the ports to relevant ones

2. Allow filtering ports by name (same as display of waypoints)

I think this is easy to do and I would be happy to do it.  Requires the user to know something about his port (e.g. the first few characters)

3. Display one page (e.g. 20 ports) with “Next” and “Prev” buttons to allow an infinite number of ports to be displayed. 

Easy to implement but a bit laborious to get to your port if it is, say, USB01. Note that this is a configuration function, NOT a flight time function so this might be fine. 

I have given up on reducing the number of ports so that leaves 2 or 3. I can do either and might have a preference for 2 though I think 3 is a bit easier to program. I welcome any more ideas and I am happy to do the implementation if I believe I can (so not choice 1).

Like @cmoetzing, I want to see this done.  The current workarounds (like choosing any port then hand editing the config file) are totally unnecessary.

Comments?

lordfolken, 2023-05-31:
I think the same filter as in waypoint list would be the correct way. It lists the ports when there is a low amount of ports, and will ask you to select a filter if there are many. The implementation is already there, and if there is an update it would be considered for all use cases.

GeekGoneOld, 2023-05-31:
As always, @lordfolken, your input is very welcome and appreciated. 

That’s 2 votes for #2 “Like Waypoints”. 

I will give this a bit more time for comments (I cant start for 1.5 weeks anyway) but if 2 ends up preferred, I will gladly do the long overdue PR for review.

Amilakvari, 2023-12-17:
Hello !
I'm sorry i'm really beginner on Github...

I assembled a bluflyvario with my Kobo Clara HD with XCSOAR 7.40 this week, everything was ok except this problem of serial port ttymxc0 not showing in list.
I tried to modified the appropriate .prf file XCSoarData/default.prf,  edit the port to the name ttymxc0 then save. But it's still not worked.
This is a christmas gift for my wife, and I'm lost...Thanks for your help.

GeekGoneOld, 2023-12-17:
> I tried to modified the appropriate .prf file XCSoarData/default.prf, edit the port to the name ttymxc0 then save. But it's still not worked. This is a christmas gift for my wife, and I'm lost...Thanks for your help.

Sorry I still haven't gotten around to filtering the ports.  The workaround, however, should work perfectly.

So here are the steps:

1. configure your device (BlueFly) in XCS (sim) using any unused tty port (e.g. ttya4) and remember the port
2. quit XCS, poweroff the Kobo then power it on to get to the splash screen (fly/sim)
3. in the network menu turn on WiFi and when it is on, click on Telnet server.
4. telnet into the device (using PuTTY over wifi) and verify the port physical port exists
    type ls /dev/ttym*

    Verify that you see ttymxc0.
5. again in telnet edit default.prf (or whatever .prf you use)
    type vi /mnt/onboard/XCSoarData/default.prf
    type :%s/ttya4/ttymxc0/
    type :wq
6. still in telnet, reopen default.prf (or whatever .prf you use) to verify the edit took place
    type vi /mnt/onboard/XCSoarData/default.prf
    type :/ttymx

     you should see that the tty port is ttymxc0
7. poweroff the kobo and restart it
. go into XCS (sim) and verify the port is now ttymxc0

Let me know if it doesn't work right for you.  We will get you past this problem before Christmas!

Amilakvari, 2023-12-19:
Hi !!!

After an full evening of "WTF, I will wash with benzin and dry with a lighter", it's seems it's working ! 

Finally, I have to redo on both side, fly and sim. I think that the problem came from my first modification directly on the microSD in the default.prf with Linux.

I will test it tomorrow outside, I have a metal top roof which probably block the GPS.

Thanks a lot !

closed 2023-12-21 by MaxKellermann

GeekGoneOld, 2023-12-23:
Fantastic. Great present!

PR #891 (2022-06-28): Filter pseudo terminals from serial port list
Brief summary of the changes
----------------------------
Set a higher limit for the enumeration list.
This makes the list longer but at least the important ports entries show up again.

Related issues and discussions
------------------------------
Closes #888