XCSoar/XCSoar #892: docker android build

volkmarnissen, 2022-03-21:
XCSoar versions having and not having the problem
-------------------------------------------------
XCSoar main branch

Expected behavior
-----------------
I wanted to do a docker build for android and followed the documentation in https://github.com/XCSoar/XCSoar/tree/master/ide/docker
I was expecting to get a build without any configuration.

First approach:
```
docker run \
    --mount type=bind,source="$(pwd)",target=/opt/xcsoar \
    -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest xcsoar-compile ANDROID
```

Actual behavior
---------------
The build failed, because the android sdk and ndk were not present at all.

Second approach:
```
docker run \
    --mount type=bind,source="$(pwd)",target=/opt/xcsoar \
    -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest /bin/bash
```
I found out, that 
```
 /root/install-debian-packages.sh and
 /root/install-android-tools.sh
```
were not executed. Even if this is stated in https://github.com/XCSoar/XCSoar/blob/master/ide/docker/Dockerfile
Additionally, I had to call
```
git submodule init
git submodule update
```

Adding the following lines to the build/local-config.mk fixed the rest:
```
ANDROID_SDK := /root/opt/android-sdk-linux
ANDROID_NDK := /root/opt/android-ndk-r23b
ANDROID_BUILD_TOOLS_DIR = $(ANDROID_SDK)/build-tools/29.0.3
ANDROID_SDK_PLATFORM =android-30
```
Now, the packages were built nicely.
So, the docker build feature is great. It is also fast.:+1:
I have no experience with make or docker, so I don't know how to fix this issue on my own.

This issue can be used as a workaround for the problem.

Steps to reproduce the behavior
-------------------------------
Use ubuntu impish as docker host and proceeded as described in 
https://github.com/XCSoar/XCSoar/tree/master/ide/docker
to call an ANDROID build

Do you have any idea what may have caused this?
-----------------------------------------------
Yes. See above.

Do you have an idea how to solve the issue?
-------------------------------------------
No. :slightly_frowning_face:
But if someone  points me to the right direction, I can try to fix it.

lordfolken, 2022-03-21:
Please post error messages you are getting. 

This container is part of our build pipeline and this built an android package 2 days ago. See:
https://github.com/XCSoar/XCSoar/actions/runs/2008414338

The most common mistake I have had is that people do not use --recursive (submodules) option when checking out the code initially. 

https://github.com/XCSoar/XCSoar#getting-the-source

volkmarnissen, 2022-03-21:
Thank you. the latest changes in targets.mk where not in my feature branch.That caused the issue.
Sorry, I could have checked that first.

Yes, I see, the issues seem to be fixed. 
I'll update the build.rst.

volkmarnissen, 2022-03-22:
The following information are currently missing or not easy to find in doc/*.rst:

Best Practise: 

- Which tools are most commonly used for development?
  - E.g. docker for all non unix builds, local build for unix builds, eclipse for C++ development. Android Studio?
  - Performance of build for the individual build set up
    - Compare build setups
    - Improve makes
- How to set up docker?
  - Mainly content from ide/docker/readme.md
  - How to set up permissions?
  - How to check "sanity" of opt/XCSoar directory?
- How to work with eclipse?
  - How to set up eclipse for XCSoar? (covered in existing documentation
  - How to avoid build problem in eclipse UI?
- How to work with Android Studio?
- How and when test mulitple platforms?
- How to work with git/ github
  - How to clone a repository?
  - How to rebase from XCSoar/XCSoar?
  - How to merge?
  - How to push a bugfix?
  - How to push a new feature?

Is this list complete?
If yes, I will set up a document structure to cover the topics. I will also add the existing documentation into this structure and write new documentation as far as I can. However, there will be gaps where my knowledge is missing.
I will create a new branch "devDocu" for this topic.
Then it should be reviewed and completed.

elgandoz, 2022-05-04:
> So im able to build Android with these commands:
> 
> ```shell
> git clone --recursive https://github.com/XCSoar/XCSoar  
> cd XCSoar
> sudo docker run     --mount type=bind,source="$(pwd)",target=/opt/xcsoar     -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest xcsoar-compile ANDROID
> ```

Hello folks, I'm currently trying to compile for Android with Docker 4.7.1 on MacOS 12.3.1 (Monterey) but the process gets stuck, and it doesn't advance after few hours running. Here's my one liner:
```bash
git clone --recursive https://github.com/XCSoar/XCSoar; cd XCSoar; docker run --mount type=bind,source="$(pwd)",target=/opt/xcsoar -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest xcsoar-compile ANDROID
```
I'm trying on 2 different Macs, one with the M1 (amd64) and the other is the old Intel. I get slightly different results, but the process hangs more or less near the same part:
| M1 | Intel |
|:--:|:--:|
| [image] | [image] |

NB: on the M1, I'm adding the parameter `--platform linux/amd64` as indicated somewhere on GitHub to suppress a warning:
```
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
```
Either way it gets stuck at the same point.

Compiling for UNIX works just fine.

Any idea on what could be the issue?

volkmarnissen, 2022-05-04:
Hi,

I faced the same issue. Reason: It seems as if android.mk uses different different SDK/NDK  than the on installed on docker image. My local build runs with this script:
```
#!/bin/bash
ANDROID_NDK_API=23
ANDROID_SDK=~/Android/Sdk
ANDROID_NDK_ROOT=$ANDROID_SDK/ndk
ANDROID_SDK_PLATFORM=android-30
echo $ANDROID_SDK
ANDROID_BUILD_TOOLS_DIR=$(ls -d $ANDROID_SDK/build-tools/30* | grep -v rc | tail -1)

ANDROID_NDK=$( ls -d $ANDROID_NDK_ROOT/$ANDROID_NDK_API*)

if [ ! -d "$ANDROID_SDK/platforms/$ANDROID_SDK_PLATFORM" ]; then
        echo "Directory $ANDROID_SDK/platforms/$ANDROID_SDK_PLATFORM does not exist"
	exit 2
fi
if [ ! -d "$ANDROID_NDK" ]; then
        echo "Directory $ANDROID_NDK does not exist"
	exit 2
fi
echo make -j22 ANDROID_NDK=$ANDROID_NDK ANDROID_SDK_PLATFORM=$ANDROID_SDK_PLATFORM ANDROID_NDK_API=$ANDROID_NDK_API ANDROID_BUILD_TOOLS_DIR=$ANDROID_BUILD_TOOLS_DIR ANDROID_SDK=$ANDROID_SDK $*
make -j22 ANDROID_NDK=$ANDROID_NDK ANDROID_SDK_PLATFORM=$ANDROID_SDK_PLATFORM ANDROID_NDK_API=$ANDROID_NDK_API ANDROID_BUILD_TOOLS_DIR=$ANDROID_BUILD_TOOLS_DIR ANDROID_SDK=$ANDROID_SDK $*
```
May be this helps.

elgandoz, 2022-05-04:
Hey @volkmarnissen, thanks for the quick help.
I made a script with your code, within the container, but I get `Directory /root/Android/Sdk/platforms/android-30 does not exist`.
Do i need to install the SDK manually?

volkmarnissen, 2022-05-04:
Yes, I installed Android Studio and then downloaded SDK and NDK of a matching version.
BTW It is not required to run the make command with root permission.

elgandoz, 2022-05-04:
I'm running the script from within the container shell, where you log as root.
So I assume you installed the S/NDK in you host environment, but I didn't want to do that...
I thought the point of docker was to have a consistent working environment without external dependencies.

volkmarnissen, 2022-05-04:
Then you can try to find the SDK and NDK there and change the directories in the script.

volkmarnissen, 2022-05-04:
You are right, this is the idea of docker, but there is a depenency of the docker image to the makefile.
So, if this is not in sync, as it is now, it is an issue.
I found it easier to work locally for android, unix, kobo and windows. I don't have a solution vor raspberry pi.

lordfolken, 2022-05-04:
The master branch builds without problems without any external dependencies and no modifications. 
The image is used always in the pipeline on every commit, and builds pretty much all architectures. You can see this here:
https://github.com/XCSoar/XCSoar/runs/6289228142?check_suite_focus=true

elgandoz, 2022-05-04:
@lordfolken I'm sure it works there, but following the guide, starting from 0 and by copying and pasting the command mentioned above, it doesn't compile on my 2 MacOS machines, plus @volkmarnissen  one.
Checking the log you posted above I can see there are several references to ` /root/opt/android-ndk-r24/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++`, but one of the first messages I get is:
```
make: /root/opt/android-ndk-r23b/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++: No such file or directory
```
I'm not an expert at all of Android development, but how come we have 2 different versions of the ndk (24 vs 23b)?
[image]

OT: are the artifacts uploaded on each commit publicly available?somewhere

lordfolken, 2022-05-04:
Maybe you need to run the container as your UID with --user $UID
That way the files will correctly belong. Be sure to delete the output folder beforehand. (or make clean on the host)

elgandoz, 2022-05-07:
Follow up, concerning **compiling via docker on MacOS**
As I found out some time ago the volume where the compilation occur must be Case Sensitive (see #712 ), which is not MacOS default (this wasn't the issue).

**On Intel architecture**
The compile process gets stuck while downloading the thirdparty libraries, listed in [./build/thirdparty.py](https://github.com/XCSoar/XCSoar/blob/master/build/thirdparty.py#L154) and defined in [./build/python/build/libs.py](https://github.com/XCSoar/XCSoar/blob/master/build/python/build/libs.py#L134).

The same code handles the [download of the boost library](https://github.com/XCSoar/XCSoar/blob/master/build/libboost.mk#L14) ok, but it gets stuck on the [`__download()` function](https://github.com/XCSoar/XCSoar/blob/master/build/python/build/download.py#L11): the process hangs there, not throwing any exception or not trying the alternate URL. In the folder I can observes the creation of the temp file with zero bytes (not the lock. file). I tried changing the order of the libraries, to test different URLS, with the same result, stopping on the first occurrence of the 3rd party libs.

The only way I could find to stop the process is to restart docker, `Ctrl+C` or stopping the container has no effect.

Checking the permission of the downloaded files, the ownership is correct, since docker already runs with my UID, but the permissions are slightly different (600 instead of 644):
[image]
Could that be the issue?

As dirty workaround, I downloaded all the libs manually and placed them in `./output/download`, it then compiles without problem when running `make TARGET=ANDROID`.
If I run `xcsoar-compile ANDROID`, which uses the target `ANDROIDFAT`, for some reason, which I do not understand, [it changes something in the 3rd party libs](https://github.com/XCSoar/XCSoar/blob/master/build/thirdparty.mk#L4-L11) while compiling, resulting in the process to hang as described before.

**On M1 architecture**
Unfortunately I didn't manage to get a build.  The problem seems different but it stops in near the same part, visible in on the screenshots above (`Traceback: [...] File "/opt/xcsoar/build/python/build/download.py", line 40, in download_and_verify" os.link(tmp.name, path) FileExistsError: [Errno17] File exists: [...]`).
The error occur regardless if I download I place the libs manually or not.
Additionally, when the libs are placed I get an extra `qemu-x86_64: unable to find CPU model 'Deep Thought'`.

elgandoz, 2022-05-07:
@lordfolken, are the artifacts uploaded by GitHub Action publicly available? It would be handy for testing some of the open PRs...

lordfolken, 2022-05-30:
@elgandoz yes, under the actions, select the appropriate build, and you can download the artifact from there.