XCSoar/XCSoar #893: Docker build for TARGET=PI

volkmarnissen, 2022-03-21:
XCSoar version
--------------
XCSoar main branch

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Even when following the developer manual 7.0 and reading in the forums I didn't succeed. I was not able to build XCSoar for PI
`make TARGET=PI`.
One of the open issues is how to get a proper sys root.
A copy of the `/usr/include` and `/usr/lib` of an existing raspberry pi as sysroot did not succeed.
Partly, make still used the wrong includes.

The current docker image has support for a lot of makes already. If the build for PI was added, it can be very simple to build them all.
Alternatively or additionally, it would be helpful to get a tested description how to set up the target correctly.

May be the docker image can write a `build/docker-config.mk` which contains all settings specific to the docker image build. 
This file can be included by docker make runs. The local-config.mk should be ignored.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
I was trying to build the raspberry pi application.
The developer documentation didn't help completely. (May be because I am not so experienced with multi platform unix builds).
The development tool states that a sys root is required. But it doesn't state what it is or how to one can get it.
The forum is also not helpful in this case.
Other forum users had the same issue. Google search was also not so helpful here.

So, I'd like to have a simple way to build the PI target. Docker would be my first choice.

cmoetzing, 2022-12-23:
Do you expect the container to run on x86_64 hardware and produce a PI binary or do you want to run the container on a PI itself?

volkmarnissen, 2022-12-23:
I expect the container to run on x86_64 hardware like the other containers.

Volkmar