XCSoar/XCSoar #894: Android 11 "XCSoarData" is no more visible

volkmarnissen, 2022-03-23:
XCSoar versions having and not having the problem
-------------------------------------------------
XCSoar main branch

Expected behavior
-----------------
If you want to add files to the XCSoarData directory in Android, you need to access it.
This is relevant for cup file, lua files ...

Actual behavior
---------------
If you use a android based File Manager, it will show the `/Android/data` directory as an empty directory. This happens  even after starting up a new installation.

As a workaround, you can connect the device via USB to a windows based system. You will see the directories and can edit them.

Steps to reproduce the behavior
-------------------------------

- Use a Samsung Galaxy Tab A7 (Most likely every other android 11 based tablet will do also).
- Install XCSoar.
- Run it for the first time.
- Open a Filemanager
- Navigate to the XCSoarData directory.
- The directory is not visible.

Do you have any idea what may have caused this?
-----------------------------------------------
A change to the android operating system.

Do you have an idea how to solve the issue?
-------------------------------------------
I see two options:

1. Move the XCSoarData Directory to the Documents diretory (like in Windows based installations). 
**Disadvantage:** The large xcm files won't get deleted when removing the app)
**Advantage:** Removal and readding the app is easier. (This scenario is only relevant for development).
2. Have two XCSoarData Directories `/Android/...` and `../Documents/XCSoarData`
**Disadvantage:** It might come to confusion if there are duplicate files.
**Advantage:** Even after removal of the app, the added files will persist.

My proposal: The second option is the better choice.(Or a better idea)

MaxKellermann, 2022-03-23:
Sorry, we can't do anything and we have no choice. This change was pressed by Google upon all applications. XCSoar uses the path you see in the system configuration, and has no other choice.

volkmarnissen, 2022-03-23:
Sure, no problem.
According to the [technical documentation ](https://developer.android.com/training/data-storage/shared/documents-files) a solution is possible.
So, I assume, it's a matter of man power and priority. Am I right?

volkmarnissen, 2022-03-23:
A third option:
Just read files created by users from the documents directory:

- lua files, 
- checklists, 
- additional waypoint files
- Additional tasks?

All of these files will be created by the user. XCSoar won't modify them.
Duplicate files are unlikely.

MaxKellermann, 2022-03-23:
All this API does is give apps a non-standard way of accessing certain files. This does not give file permissions; POSIX file access is not possible; it provides a Java-only which internally uses IPC with another app to actually do file access. This would require rewriting all parts of XCSoar which handle files (and that rewrite would only work on Android).

volkmarnissen, 2022-03-23:
I see. Too bad. So, it's possible, but a really huge effort.
I will keep that in mind. May be I'll try to implement it later.

MaxKellermann, 2022-03-23:
I doubt I would merge such a "feature" into XCSoar. This would add huge amounts of code that somebody would need to keep maintaining forever, and I certainly won't.
If you have hundreds of hours of free time for XCSoar development (which you would need to implement this), there are many things you'd rather do that are more helpful for XCSoar users.

lordfolken, 2022-03-23:
I think it would be wiser to make flight logs more accessible. I.e via share or sync functionality. 

And maybe a share / save / export possibility for the settings. 

The rest is data that can be redownloaded.

volkmarnissen, 2022-03-23:
Flight logs are certainly to be considered when working on it.

> The rest is data that can be redownloaded.
This is only true for non user defined files. It is not true for lua files, checklists ...

I still keep the complete topic in my mind.
May be we can define a simple C++ based interface to exchange small text files from one defined directory  with no subdirectories.

Something like
TCHAR *[] UserStore.listFiles( "lua");
TCHAR *UserStore.getFile("lua", "init.lua");
int UserStore.putFile("igc", "123.igc", theIgcFileContent, overwrite)

int UserStore.putFile(".", "default.prf", theProFileContent, overwrite);

One implementation uses POSIX. One implementation uses android java.
This interface can then be used only where it's required.

However, this is nothing, I'd like to work on now. Let's continue the discussion, when I finished the other topics.

What do you think about reopen the topic and assign myself?

MaxKellermann, 2022-03-23:
I'd really like you to spend the time elsewhere, but in the end, it's up to you how you spend your free time.

volkmarnissen, 2022-03-23:
Thank you for reopening. 
I'd like to focus on usability for new users. However, I appreciate the development you do in the newer areas.
As of now, I am a new user, so it's easy for me to experience the issues.
May be this changes over time. Then I will close the topic with or without a solution and work on other topics.

lordfolken, 2022-04-02:
I think the main thing we should solve here, is the igc logs. A log viewer with an android share intent would probably alleviate the situation. And maybe another one on the configuration page, that opens the folder in an android file viewer.

lordfolken, 2022-04-26:
I would propose an android intent button on the profile path in the settings dialog. That way the directory that is in actual use could be opened with an android file manager.

lordfolken, 2022-04-26:
For other platforms the button could call xdg-open with RunFile
https://github.com/XCSoar/XCSoar/blob/07746f16fbed24d0d0cf31d8fa18b943896e32c1/src/system/RunFile.cpp

volkmarnissen, 2022-04-26:
Meanwhile I found out, that, there is a file browser which can access the app/data directory.
So, it' s really needed only for igc files for OLC (because OLC has no API). OLC is not so relevant for me. So, I am no longer sure, whether I want to work on it.
Currently, I am struggling with setting up a debuggable android version.
I'd like to using docker build and android studio.

lordfolken, 2022-04-26:
a plan of mine would be to extend the vagrant box to a full environment for development. 
android-studio is available as a snap package https://snapcraft.io/android-studio

volkmarnissen, 2022-04-26:
It was no issue to setup AndroidStudio. It is also no problem to import XCSoar-debug.apk into it and to deploy and run it.
Vagrant is nice, but I am not sure whether it helps with my issue.

I was struggling with more basic things. :-)
What is the docker command line to generate a debuggable apk?

It took me two days to find out, whats wrong, because I am new to android development.
First, I tried to set up a gradle build. It was buildable and debuggable, but the resource file structure was different.

Now, I know, that the XCSoar-debug.apk is not debuggable, because it has  no  android:debuggable="true" in it's AndroidManifest.xml. This is hidden in the gradle build, but it seems to be missing in the android.mk

My proposals:
1. Use xslt in android.mk to generate a AndroidManifest.xml based on settings in options.mk.
This approach allows also to set minSdk and other settings from Makefile variables. There will be only one AndroidManifest.xml in github.
In my ubuntu environment, there is a xslt command line processor named xsltproc. This can be used to do the stuff.

2. Use gradle build for android. Only the generation of the C++ library should be built in exiting make. 
Android build seems to be a moving target. Gradle might lower the migration effort when migrating to new versions.
Additionally, it might be more convinient to use Android Studio for the java part.
However, this requires a change of the resources structure of XCSoar.

MaxKellermann, 2022-04-26:
What's the point of Gradle? It's a heavily bloated, extremely slow and obscure tool which needs to download megabytes of stuff from the net before it can do anything.
You have not explained how Gradle helps with your problem.
Building XCSoar is just a matter of typing `make TARGET=ANDROID`.

volkmarnissen, 2022-04-26:
> What's the point of Gradle? 

gradle is integrated to android studio and has an integration to android builds. It's maintained by google.
I don't insist on it. :-)

> Building XCSoar is just a matter of typing make TARGET=ANDROID.

Yes, but the version built  "XCSoar-debug.apk" is not debuggable, because of the missing entry attribute android:debuggable="true" in AndroidManifest.xml. So, either the name XCSoar-debug.apk is missleading or it's an issue.

The following style sheet:

```<?xml version="1.0"?>
<xsl:stylesheet version="1.0" xmlns:xsl="http://www.w3.org/1999/XSL/Transform" xmlns:android="http://schemas.android.com/apk/res/android">
	<xsl:param name="debuggable">false</xsl:param>

  <xsl:template match="application">
  <xsl:copy>
	  <xsl:apply-templates select="@* "/>
 	  <xsl:attribute name="android:debuggable" ><xsl:value-of select="$debuggable"/> </xsl:attribute>
 	  <xsl:apply-templates select="node()"/>
 </xsl:copy>
  
  </xsl:template>
  <xsl:template match="*">
  <xsl:copy>
 	  <xsl:apply-templates select="@*"/>
 	  <xsl:apply-templates select="node()"/>
 </xsl:copy>
  </xsl:template>
  
 <xsl:template match="@*">
  <xsl:copy/>
  </xsl:template> 
 <xsl:template match="@android:debuggable">
  </xsl:template> 
</xsl:stylesheet>
```

together with this makefile snippet
```
OUTMANIFEST = $(ANDROID_OUTPUT_DIR)/manifest/AndroidManifest.xml 
XSLFILE = android/AndroidManifest.xsl
XSLT = xsltproc
$(OUTMANIFEST) : $(MANIFEST) $(XSLFILE) | $(ANDROID_OUTPUT_DIR)/manifest/dirstamp
	@$(NQ)echo "  XSLT"
	$(Q)$(XSLT) --stringparam debuggable true --output $(OUTMANIFEST) $(XSLFILE) $(MANIFEST) 
```
 can generate the attribute dynamically.
It can also update other attributes if needed. E.g.- minSdk, version Number ...
The approach is similar to the gradle implementation

MaxKellermann, 2022-04-26:
> Yes, but the version built "XCSoar-debug.apk" is not debuggable, because of the missing entry attribute android:debuggable="true" in AndroidManifest.xml.

Then .... add it!

> So, either the name XCSoar-debug.apk is missleading or it's an issue.

It's only misleading because "debug build" can mean different things. It can mean:
- compiler optimizations disabled
- no (script) minification
- no obfuscation
- assertions and other runtime checks enabled
- debug symbols not stripped
- (did I forget something?)

It means some of the above things for XCSoar.
For you, it means the Android-proprietary "debuggable" flag is enabled. It's not that currently, so you find it misleading.

> The following style sheet:

What the ....
Did you really just post 24 lines of obscure XSLT just to add a single XML attribute, instead of just submitting a single-line PR to add this XML attribute?

> can generate the attribute dynamically.

Why dynamically? If this flag is useful, what's the point of disabling it at all?

On my operating system of choice (which is not Android), all processes are "debuggable", of course they are! Being able to hack and debug everything is what makes Linux so fancy for me. Android has the Linux kernel, but its spirit is not Linux; Android takes away your freedoms, it locks you out of your own hardware.

I wouldn't want to take any action which takes away more rights from XCSoar users. They shall exercise all freedoms possible in that lock-in Android world.

Certainly I don't want to have obscure XSLT code whose only purpose is to take away user's freedoms. No more engineering against user's rights!

> It can also update other attributes if needed. E.g.- minSdk, version Number ... The approach is similar to the gradle implementation

Sure the version number should be assigned automatically - which I could do with a simple `sed` half-liner, just didn't bother yet.

But why would I want to update minSdk dynamically?

volkmarnissen, 2022-04-26:
> Then .... add it!

Now, I can add it. After two days of investigation. It would be better if it is clearer how to debug XCSoar.
It's not like other android studio based apps. There are differences. One has to know how to deal with it.

> Why dynamically?

I am not sure whether it is possible to upload to play store with debuggable=true. 
I will add it to android/testing/AndroidManifest.xml and change TESTING in options.mk.(locally)

> On my operating system of choice (which is not Android), 

Same with me. However, when I am flying, I use android and I found out, that there are gaps in functionality in android (The focus doesn't work as on other operating systems).
So, I'd like to debug it. It happens only on android.

> But why would I want to update minSdk dynamically?

You are right. That's not needed.

OK. I'll leave the make file as it is and update the developer manual: I'll describe how to debug an android application.
May be other new  developer might find this useful.

closed 2022-05-03 by MaxKellermann