XCSoar/XCSoar #896: UTCOffsetSigned not working for negative offsets

heyleehenry, 2022-03-29:
XCSoar versions having and not having the problem
-------------------------------------------------

v7.19-android is showing the issue.  I am a new user of XCSoar and this is the first version I have used.

Expected behavior
-----------------

I set the UTC offset in Config, System, Setup, Time to -5h for Central Daylight Time.  Local time was correctly displayed.  After I quit XCSoar and restarted, local time reverted back to UTC.  Upon setting the UTC offset again, local time was correct.

Actual behavior
---------------

UTC is not getting picked up correctly from the profile.  Profile shows UTCOffsetSigned="4294949296".  That seems correct, negative 5hr * 3600 seconds and local time is correct immediately after setting.

Steps to reproduce the behavior
-------------------------------

Go to Config, system,setup, time.  Change UTCOffset to a negative hour offset.  Observe local time in an infobox.  Quit XCSoar. Start XCSoar and observe local time in an infobox.  Also view the current profile in a text editor and view UTCOffsetSigned.

Do you have any idea what may have caused this?
-----------------------------------------------

Looking at the source in Profile/ComputerProfile.cpp, the routine LoadUTCOffset calls Get with UTCOffsetSigned to get "value". If successful, it will next check the validity of "value".  Since a positive hour offset works, it would be reasonable to assume that Get(UTCOffsetSigned, value) did return successfully.  It would also seem likely that the validity check is passing for positive values, "value > 13 * 3600" is not true.  I'm guessing the negative check "value < -13 * 3600" is returning true and RoughTimeDelta is not getting called.  Possibly an issue with "int value" and the "negative" UTCOffsetSigned="4294949296".

Do you have an idea how to solve the issue?
-------------------------------------------

I would check the value of the variable "value" after the call to map.Get and make sure it matches the value in the Profile.  If it was correct I would next want to know if the validity check was true and returning FALSE for the LoadUTCOffset routine.

heyleehenry, 2022-03-29:
> Hi, thx for the report.
> Unfortunately i cannot reproduce this on Android v7.23 or on current master unix.
> Is use GPS Time setting on ? (expert option)
> You can get latest releases from https://download.xcsoar.org/releases/ <https://download.xcsoar.org/releases/>
> (unfortunately play store is against us due to privacy issues.)

I do have Use GPS Time on.  I will try it on 7.23.  Thanks.

heyleehenry, 2022-03-29:
I updated to v7.23.  It now seems to handle the UTCOffsetSigned better.  But whatever offset I choose is changed to -6h after I quit and restart XCSoar.  Both with “Use GPS” on and off.  Could it be picking up my location and be looking for a daylight savings time value?

lordfolken, 2022-03-29:
can you stat in SIM mode, and check if that changes the behavior for you?

lordfolken, 2022-05-02:
as no reaction closing.

closed 2022-05-02 by lordfolken

heyleehenry, 2022-10-11:
I get the same in SIM.  When I change the UTCOffset and the local time is correct, I quit XCSoar.  Then I look at the profile and the UTCOffsetSigned value looks correct.  When I start back up, SIM or FLY, the offset in Config/System/Setup/Time has been changed to -6.  Looking at the profile, it still has the value for -5h.  So it does not appear that the offset in the profile has changed but the configuration is displayed as -6 no matter what I enter.  I tried -4h thinking it might be getting off by one but the offset after restarting was still -6.

UTC Offset of zero or positive works correctly for me.  Only negative offsets result in -6h after restart.