XCSoar/XCSoar #897: git pull error: unauthenticated git protocol

fraktal-sb, 2022-04-02:
XCSoar versions having and not having the problem
-------------------------------------------------

all

Expected behavior
-----------------

"git pull" should update the local software without error message

Actual behavior
---------------

"git pull" produces the following messages:

fatal: Fehler am anderen Ende:
The unauthenticated git protocol on port 9418 is no longer supported.
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.

Steps to reproduce the behavior
-------------------------------

use "git pull" on Debian 11

Do you have any idea what may have caused this?
-----------------------------------------------

Update of .XCSoar/.git/config necessary in line 7

from
url = git://github.com/XCSoar/XCSoar
to
url = https://github.com/XCSoXCSoar/.gitar/XCSoar

Do you have an idea how to solve the issue?
-------------------------------------------

new git policy (see error message/URL);

see also

Openvario/meta-openvario#287

lordfolken, 2022-04-02:
Well it still works, but you need a github accounwith ssh keypair.

fraktal-sb, 2022-04-02:
what is the best long term solution?

Update the config file or github account with keypair and how should the github account be configured and used?

closed 2022-04-02 by lordfolken