XCSoar/XCSoar #899: Menu Nav/Waypoint List do not display any  waypoints

MikeBravo15, 2022-04-04:
XCSoar versions having and not having the problem
-------------------------------------------------

Latest official 7.23/ latest development build (7.24), Android, Linux, PC

Expected behavior
-----------------

When I open menu Nav/Waypoint List, all waypoints of waypoints files, selected via menu System/Site Files will be displayed, Then I can filter them via filter tools on the left side of the screen.

Actual behavior
---------------

No waypoints are displayed, although all filters are set to display all items (*).

Steps to reproduce the behavior
-------------------------------

Open menu Nav/Waypoint List

Do you have any idea what may have caused this?
-----------------------------------------------

No

Do you have an idea how to solve the issue?
-------------------------------------------

Unfortunatelly not

lordfolken, 2022-04-04:
Whatis your waypoint source?
Can you attach a waypoint file you are using?
What are your settings onside the files Dialog?

MikeBravo15, 2022-04-04:
I know, that setting the filters will display requested waypoints. Is this behavior intended (long list not displayed) or a bug?

lordfolken, 2022-04-04:
https://github.com/XCSoar/XCSoar/blob/50f73680eae11654cf3f0134c18e11c8986d81f8/src/Dialogs/Waypoint/WaypointList.cpp#L268-L275

There is a limit of 500 waypoints. if your result is greater than that, nothing is displayed. 

The commit (12 Years ago!) https://github.com/XCSoar/XCSoar/commit/5f5b189e45940aad854d62ac442711b93e624cbf 
states its for performance reasons, as each waypoints's distance/direction is calculated to the present position. 

So not a bug. Works as intended.

MikeBravo15, 2022-04-05:
OK, thanks for info. Would you consider to raise the number to ie 800/1000? We have come some way performance wise

lordfolken, 2022-04-05:
In the end there needs to be a cap in order to protect operational stability. 
Its also human performance factor. The scroll bar's drag element gets really small when you add more entries.
Also i don't think displaying more waypoints is the answer. 

The question is really how can the project offer a better more efficient selection of waypoints, that works on touch/stick/keyboard, and resolutions from 320x200 to larger than 1024 pixels.

MikeBravo15, 2022-04-05:
OK, please close it now

closed 2022-04-21 by lordfolken