XCSoar/XCSoar #900: FallbackElevation fails at startup - waypoints w/o explicit Elevation are not displayed

bomilkar, 2022-04-06:
XCSoar version
--------------

7.23

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

When reading a CUP formatted waypoint file with an empty Elevation field, it should get the Elevation from the terrain.
In fact, this is what FallbackElevation does when the terrain information is available. But it isn't at startup.

Some time between 7.12 and 7.18 the terrain reader was put into a separate thread (??). As a result, when XCSoar starts, the terrain information is not available when the waypoint file(s) are read. Therefore waypoints without elevation information are **not** loaded.

Changing the waypoint file configuration when XCSoar is fully up an running, triggers a reload of all the waypoint files. At that stage the terrain information is available and FallbackElevation can do it's job: it supplies the elevation information and the waypoint is loaded and displayed.

I'm referring to `bool FallbackElevation(Waypoint &waypoint) const;`

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

There are good or bad reasons for having an empty Elevation field in a CUP file: a turnpoint might not necessarily need it; an outlanding field might not come with reliable elevation information; etc. In these cases, XCSoar should read the elevation value from the terrain information, as it used to do when it completed reading the terrain **before** the waypoints.

A possible solution could be to remember which waypoints requested the elevation information from the terrain and (re)process them **after** the terrain is completely loaded. That behavior should be documented in the user guide. NB: SeeYou's CUP file format description, dated 2018-06-07, doesn't talk about empty elevation fields.

I'm not suggesting to use bogus elevation values  if the terrain is unavailable!

bomilkar, 2024-01-22:
Just wanted to reconfirm: this issue still exists in the recent build (v7.42-Linux~git#010af8a728).
I think I know what happens, I already described it above. Basically, FallbackElevation doesn't have the topography from the map file because the map file hasn't been loaded at this point.
But I have no idea how to fix the issue.

lordfolken, 2024-01-29:
> But I have no idea how to fix the issue.

This LoadWaypoints function needs to be somehow called after LoadTerrain has completed, if it was executed at runtime. 

https://github.com/XCSoar/XCSoar/blob/010af8a728a6230d98e33e5376ef963f36a2a1a7/src/Startup.cpp#L427-L434

lordfolken, 2024-01-30:
Can we set a flag when FallbackElevation had failed. If that flag is set after LoadTerrain completes, then run LoadWaypoints again?

The waypoints are already read at startup, so we don't wait for the terrain, for waypoints to be available. 

Once terrain completes, we can rerun the waypoints initialization and recalulate the missing altitudes. 
Because in theory you could load a different terrain where the fallback calculation results in a different altitude. So its not enough to just recalculate the missing points.

Also a question that comes to mind, is what happens to current active waypoints, such as in a task or goto?

bomilkar, 2024-01-31:
LoadTerrain runs asynchronously. Right?
How can Startup know it's done?

MaxKellermann, 2024-01-31:
> LoadTerrain runs asynchronously. Right? How can Startup know it's done?

`AsyncTerrainOverviewLoader` signals a `UI::Notify` when it's done, and that's `MainWindow::terrain_loader_notify`, and its callback function is `MainWindow::OnTerrainLoaded()`.

bomilkar, 2024-02-01:
@lordfolken : I have a first cut at it. It does address the issue, it doesn't seem to have any negative side effects.
Shall I submit a PR so you can look at it? Or how can we proceed.
It's a proof of concept and I really would like to get your thoughts on it.

lordfolken, 2024-02-01:
@bomikar push your branch to your repo. I can have a first look. Ultimately though Max has to approve it.

bomilkar, 2024-02-02:
Here it is: https://github.com/bomilkar/XCSoar/blob/master/src/Startup.cpp

I really appreciate your feedback, @lordfolken . I know it's far from approval!
Ideally I would like to (try to) fix each waypoint's elevation in line 236, but I haven't figured out how to access FallbackElevation() at that point. That would minimize the compute effort.

lordfolken, 2024-02-02:
@bomilkar 
While you are in the right place, you have to reload the waypoints; all of them, and let 
FallbackElevation do its work in the Waypoints Reader.
With this we would add an additional code path of when and how the waypoints get their elevation: 

If user starts with one terrain, and then loads a different terrain (for example_HighRes) the elevation doesn't get reevaluated for that new terrain. Potentially resulting in a different elevation for the same waypoint, the next time the user starts xcsoar.

bomilkar, 2024-02-02:
> If user starts with one terrain, and then loads a different terrain

Very good point. I didn't think of this.
However, only those waypoints that used FallbackElevation need to reevaluate for the (new) terrain. Adding "bool elevation_from_terrain" to struct Waypoint to remember if FallbackElevation was called (successfully or not).
I expect most if not all Waypoints to get their elevation from the CUP file. Reevaluating those would yield the same result.

lordfolken, 2024-02-02:
It also occurred to me, that a source flag for the waypoint's elevation would be a way of identifying those points that have been evaluated. But this is again more code and complexity. I'd rather only have one code path.

bomilkar, 2024-02-02:
Is this what you mean: 
https://github.com/bomilkar/XCSoar/commit/2cbbfc95de2e2088d1d13c3510bd5560d163eaa4

MaxKellermann, 2024-02-02:
I do think each waypoint should be able remember how the elevation was determined, and every time new terrain gets loaded, all waypoints which have terrain elevation (or no elevation) should attempt to (re)load terrain elevation. This is some complexity, but it's totally worth it.

MaxKellermann, 2024-02-02:
There's
```
bool has_elevation = false;
```
Replace it with:
```
enum class ElevationSource : uint_least8_t {
  NONE,
  FILE,
  TERRAIN,
} elevation_source = ElevationSource::NONE;
```
This way, we don't even increase memory usage. Only very little code complexity added and no runtime overhead at all.

bomilkar, 2024-02-02:
This would do it:
```
  auto &way_points = *data_components->waypoints;
  for (auto &wp : way_points) {
    if (wp->elevation_source != FILE) {
      factory.FallbackElevation(wp);
    }
  }
```
if I new how to get "factory".

MaxKellermann, 2024-02-02:
Too fragile - what if somebody adds a new enum choice eventually? Then the value will get overwritten, without a compiler warning.

bomilkar, 2024-02-02:
I thought: The CUP file is always right, therefore (elevation_source == FILE) is the only safe source for elevation. Everything else is guesswork and should be reevaluated. Right?

MaxKellermann, 2024-02-02:
Everything else, including future enum choices that you cannot imagine yet?
I can imagine one: USER for values entered by the user. Do you want to discard and override those?
See. That's why your design is fragile.

bomilkar, 2024-02-03:
OK, you're right.

I have 2 questions:
1. which is the instance of class WaypointFactory so that I can call FallbackElevation() ?
2. if we wrap that code into a function Terrain2Waypoints(). Should this function be a member of WaypointFactory ?

closed 2026-10-12 by lordfolken