XCSoar/XCSoar #901: WndForm runs into assertion in Windows when closing with ESC key (keyboard)

volkmarnissen, 2022-04-08:
XCSoar versions having and not having the problem
-------------------------------------------------
XCSoar main branch run on windows.

Expected behavior
-----------------
Open the checklist dialog
Press ESC **key** (The button on the screen works properly)
Expectation: The dialog should close (but nothing happens)
Press the key again
You get multiple assertions because hidden windows are going to be hidden

Actual behavior
---------------
The dialog doesn't close when pressing ESC key on the keyboard
Nothing happens
If you press ESC again, you get assertions.

Steps to reproduce the behavior
-------------------------------
See above. It is reproducible.

Do you have any idea what may have caused this?
-----------------------------------------------
In the code "Form/Form.cpp", there is some #ifndef WIN_USER (line 396) part which might be part of the problem.
However, I am not so deep into it, that I can fix it.
My version of Form.cpp has a small modification, but I don't think, that it causes the issue, because it doesn't apply to the ESC key.
If you are not able to reproduce, just close the issue. I will look into it.

Do you have an idea how to solve the issue?
-------------------------------------------
No.

lordfolken, 2022-04-08:
You're right. 
Even on wine64 i can at least crash it when hitting esc.

lordfolken, 2022-05-15:
The Assertion
[image]

lordfolken, 2022-05-20:
Please test.

PR #931 (2022-05-22): Windows: Fix Assert in WndForm
fixes #901
closed 2022-05-22 by lordfolken