XCSoar/XCSoar #909: Crashes with XCSoar v7.23-Android

kitebuggy, 2022-04-20:
Firstly, this is a wonderful app.  However, I seem to be able to reproduce crashes on my Xiomi MI 6 Android device, running MIUI 10.1 Stable (effectively Android 6, I believe).

Previous versions worked a treat, but I find that when pausing the application (e.g. for swapping applications) and returning to it, the app crashes.

I can reproduce on demand and will happily supply debug logs if you could point me in the right direction for finding these?

Thanks!

lordfolken, 2022-04-20:
Hey there! 

Some android implementations kill apps in the background. So effectively once you switch to another app, XCSoar is being killed, and relaunched once selected again.

It seems that your phone features a "lock" function for background apps:
https://www.techbone.net/xiaomi/user-manual/lock-background-apps

Second thing is to ensure that xcsoar always has access to the location in the background. This can be adchived in the apps application under the location permissions.

lordfolken, 2022-04-21:
closing this. If you still have problems, reopen.

kitebuggy, 2022-04-22:
Ok, I'm not an Android expert, so this took me a long while to work out how to do this, not helped by Xiomi's weird interfaces, but please find attached an adb logcat log, which I believe illustrates the issue and may help with #883 too.

kitebuggy, 2022-04-22:
[xcsoar.log](https://github.com/XCSoar/XCSoar/files/8543101/xcsoar.log)

kitebuggy, 2022-04-22:
Please let me know if I've done this incorrectly - any guidance on how to provide a better log would be appreciated.

lordfolken, 2022-04-22:
:25.036  1805  2611 W BaseMiuiPhoneWindowManager: keyCode:82 down:true eventTime:67579 downTime:67579 policyFlags:22000002 flags:48 deviceId:6 isScreenOn:true keyguardActive:false repeatCount:0
04-22 14:43:25.155  1805  2611 W BaseMiuiPhoneWindowManager: keyCode:82 down:false eventTime:67697 downTime:67579 policyFlags:22000002 flags:48 deviceId:6 isScreenOn:true keyguardActive:false repeatCount:0
--------- beginning of crash
04-22 14:43:25.220  6639  6663 F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0x58 in tid 6663 (NativeMain)
04-22 14:43:25.334   290   290 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
04-22 14:43:25.334   290   290 F DEBUG   : Build fingerprint: 'Xiaomi/cancro_wc_lte/cancro:6.0.1/MMB29M/V10.1.1.0.MXDCNFI:user/release-keys'
04-22 14:43:25.334   290   290 F DEBUG   : Revision: '0'
04-22 14:43:25.334   290   290 F DEBUG   : ABI: 'arm'
04-22 14:43:25.334   290   290 F DEBUG   : pid: 6639, tid: 6663, name: NativeMain  >>> org.xcsoar <<<
04-22 14:43:25.335   290   290 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x58
04-22 14:43:25.367   290   290 F DEBUG   :     r0 aee15ffc  r1 00000000  r2 00000000  r3 00000002
04-22 14:43:25.367   290   290 F DEBUG   :     r4 b6858380  r5 00000000  r6 00000004  r7 00000001
04-22 14:43:25.367   290   290 F DEBUG   :     r8 aee15ffc  r9 00000000  sl 00000000  fp aee16440
04-22 14:43:25.367   290   290 F DEBUG   :     ip b684fc68  sp aee15ff0  lr b67faf11  pc b67fdcb0  cpsr a00e0030
04-22 14:43:25.371   290   290 F DEBUG   : 
04-22 14:43:25.371   290   290 F DEBUG   : backtrace:
04-22 14:43:25.371   290   290 F DEBUG   :     #00 pc 00014cb0  /system/lib/libEGL.so (eglCreateWindowSurface+47)
04-22 14:43:25.371   290   290 F DEBUG   :     #01 pc 00335bf8  /data/app/org.xcsoar-1/lib/arm/libxcsoar.so (offset 0x12b000)
04-22 14:43:25.955  1805  1882 E libprocessgroup: failed to kill 1 processes for processgroup 5847
04-22 14:43:26.074   290   290 F DEBUG   : 
04-22 14:43:26.074   290   290 F DEBUG   : Tombstone written to: /data/tombstones/tombstone_04
04-22 14:43:26.074  1805  7058 W ActivityManager:   Force finishing activity org.xcsoar/.XCSoar
04-22 14:43:26.131  1805  7058 W ActivityManager: Exception thrown during pause
04-22 14:43:26.131  1805  7058 W ActivityManager: android.os.DeadObjectException
04-22 14:43:26.131  1805  7058 W ActivityManager: 	at android.os.BinderProxy.transactNative(Native Method)
04-22 14:43:26.131  1805  7058 W ActivityManager: 	at android.os.BinderProxy.transact(Binder.java:510)
04-22 14:43:26.131  1805  7058 W ActivityManager: 	at android.app.ApplicationThreadProxy.schedulePauseActivity(ApplicationThreadNative.java:769)
04-22 14:43:26.131  1805  7058 W ActivityManager: 	at com.android.server.am.ActivityStack.startPausingLocked(ActivityStack.java:931)
04-22 14:43:26.131  1805  7058 W ActivityManager: 	at com.android.server.am.ActivityStack.finishActivityLocked(ActivityStack.java:3151)
04-22 14:43:26.131  1805  7058 W ActivityManager: 	at com.android.server.am.ActivityStack.finishTopRunningActivityLocked(ActivityStack.java:2983)
04-22 14:43:26.131  1805  7058 W ActivityManager: 	at com.android.server.am.ActivityStackSupervisor.finishTopRunningActivityLocked(ActivityStackSupervisor.java:3079)
04-22 14:43:26.131  1805  7058 W ActivityManager: 	at com.android.server.am.ActivityManagerService.handleAppCrashLocked(ActivityManagerService.java:12910)
04-22 14:43:26.131  1805  7058 W ActivityManager: 	at com.android.server.am.ActivityManagerService.makeAppCrashingLocked(ActivityManagerService.java:12806)
04-22 14:43:26.131  1805  7058 W ActivityManager: 	at com.android.server.am.ActivityManagerService.crashApplication(ActivityManagerService.java:13495)
04-22 14:43:26.131  1805  7058 W ActivityManager: 	at com.android.server.am.ActivityManagerService.handleApplicationCrashInner(ActivityManagerService.java:13002)
04-22 14:43:26.131  1805  7058 W ActivityManager: 	at com.android.server.am.NativeCrashListener$NativeCrashReporter.run(NativeCrashListener.java:86)
04-22 14:43:26.146  1805  1830 W ActivityManager: Scheduling restart of crashed service org.xcsoar/.MyService in 1000ms

W ActivityManager:   Force finishing activity

lordfolken, 2022-04-22:
Did you "lock" xcsoar like in the url i provided?

kitebuggy, 2022-04-23:
Apologies, observed behaviour is identical, but the supplied log was with it unlocked.  Here's the locked log version.

kitebuggy, 2022-04-23:
[xcsoar-locked.log](https://github.com/XCSoar/XCSoar/files/8546597/xcsoar-locked.log)

lordfolken, 2022-04-23:
There also seems some "GameBooster" service that does something to xcsoar. Can you try to disable that?
But in the end this will be very hard to debug without a device. 
@MaxKellermann could you have a look at that log? It seems to try to create a window and segfaults there.

MaxKellermann, 2022-04-23:
GameBooster. Chuckle!

This is yet another crash in the graphics driver. It is very hard to find out whether this is an XCSoar bug because of wrong use of the graphics APIs, or whether it's a graphics driverbug. Not without a device which reproduces the bug.

lordfolken, 2022-04-23:
@kitebuggy there seems to be lineageos for your device: https://forum.xda-developers.com/f/xiaomi-mi-6-roms-kernels-recoveries-other-dev.6334/ that will give a you a more recent android version (11) and new graphics drivers. I would try to install that, and use xcsoar on it.

closed 2022-05-19 by lordfolken