XCSoar/XCSoar #912: Make on-off buttons meaning more concistent

Hugal31, 2022-04-24:
XCSoar version
--------------

Version 7.20-Android (commit e0ff401c6).

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

For now, some buttons says "On" or "Off" meaning the state of a feature is respectively "Enable**d**" or "Disable**d**", like the SnailTrail mode button ([code here](https://github.com/XCSoar/XCSoar/blob/ca4be3c26eda602ab0323b75bd98164f20195680/src/Menu/ExpandMacros.cpp#L315)).
Some other says "On" meaning "Enable": a press on the button will toggle the state of the feature to "On", and vice versa, like the Airspace toggle button ([code here](https://github.com/XCSoar/XCSoar/blob/ca4be3c26eda602ab0323b75bd98164f20195680/src/Menu/ExpandMacros.cpp#L331)).

This behavior is already inconsistent, but is worse in other languages. For instance, in French, "Off" is translated ["Desactivé"](https://github.com/XCSoar/XCSoar/blob/ca4be3c26eda602ab0323b75bd98164f20195680/po/fr.po#L892) (Disable**d**) instead of "Désactiver" (Disable), which is the opposite of what the airspace toggle button mean.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

I see two solutions:

- Create two separate "On" and "Off" translation, one for the state, one of the action, like "Off_state" and "Off_action" which would be translated correctly in other languages.
- Make the behavior consistent by only having buttons giving the actual text of the feature.

I prefer the second option, but this will inverse the actual meaning of some buttons and might disturb peoples.

lordfolken, 2022-04-24:
I think there should be an indication what the current state is (i.e. classic led light on button) and the label should just add airspcae.

lordfolken, 2022-04-24:
The checkmark icon probably would work best.

lordfolken, 2022-04-25:
https://github.com/XCSoar/XCSoar/pull/913 <- please give me your opinion.

lordfolken, 2022-05-20:
There is more work in #927 please test and provide feedback.

lordfolken, 2022-07-20:
Some of the commits where cherry-picked.

closed 2022-07-20 by lordfolken