XCSoar/XCSoar #917: Frequency transmission for airspace related frequencies

abiweil, 2022-04-30:
XCSoar version: the newest one
--------------

What should XCSoar do differently, what functionality should be added?
It's a highlight of XCSoar to be able to control respectively adjust the frequency of the attched radio. It would be very useful if the feature could be extended on the airspace related frequency for example the frequency of the related Info, Radar or ATIS-frequency.
This frequency can be extracted out of the open airpace AN or AR instruction.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

lordfolken, 2022-04-30:
I agree. Missing feature and contributions welcome.
I was also thinking of a "shout at" function. Select Flarm target and switch to the frequency from the flarmnet.

MaxKellermann, 2022-08-23:
How *exactly* can the radio frequency be extracted?

lordfolken, 2022-08-23:
Extracted is the wrong word here. 

The winpilot standard has been ammend at some point to include the 
AF field 
```
*## (R) EHR 3A (Oldebroek) ###
AC R
AN EHR 3A
AH FL 185
AL 3000 MSL
AF 132.350
AG Dutch Mil
AW MON.08:00L-23:59L
AW TUE.08:00L-23:59L
AW WED.08:00L-23:59L
AW THU.08:00L-23:59L
AW FRI.08:00L-16:59L
DP 52:23:00 N 005:50:00 E
DP 52:21:30 N 005:52:30 E
DP 52:24:30 N 006:01:00 E
DP 52:27:00 N 006:00:00 E```

We already parse this field and display it as a read only field in the airspace details dialog. 
However there is currently no way to use the radio automation (like in the airfield details dialog) to send the frequency to the radio. 

Two things that I would see here:
 - Be able to send the frequency to active/standby from the airspace details dialog
 - On the airspace Warning popup dialog, add the ability to send the radio frequency to active/standby slots of the radio.

closed 2022-08-23 by MaxKellermann

abiweil, 2022-10-11:
Thank you very much for the realization of the radio-frequency-transmission out of the Open Airspace commands “AR” and “AF”
But there are some difficulties with existing open airspace files.
Obviously the information from the AR-command is omitted, if there are additional informations in the command.
For example, the command: “AR 134.505, Act. Langen INF 128.950” leads to the following result in the airspace details:
XCSoar V7.23

XCSoar V7.25

Only the command: “AR 134.505 leads to the wanted result:

At least this behaviour should be described in the manual. 
The better solution would be to display the complete info as in V7.23, if the frequency is for example the first info in the line.
I want to mention, that the special command “AF “ is denied in some task-planning-APPS like Strepla.
With kind regards
Albert Kiessling

From: Max Kellermann 
Sent: Tuesday, August 23, 2022 10:01 PM
To: XCSoar/XCSoar 
Cc: abiweil ; Author 
Subject: Re: [XCSoar/XCSoar] Frequency transmission for airspace related frequencies (Issue #917)

Closed #917 as completed via 1fa64cd.

lordfolken, 2022-10-12:
Hi Albert,

The file you have isn't according to the standard. The af field only supports one frequency, no comments or text.

http://winpilot.com/usersguide/userairspace.asp

The AR field isnt in the standard, and its origins are not clear. But generally its functionally equal to AF.

If you need additional information displayed, the only way is to add it to the AN field.