XCSoar/XCSoar #921: Flarm Targets should be icons according to aircraft type

lordfolken, 2022-05-02:
Currently all Flarm targets are the same triangle. In order to differentiate between different aircraft, the targets should be iconifed:
I.e. powerplane/glider/paraglider/helicopter. 

The should:
 * be only an outline of an icon
 * retain the color for above/below
 * have some form of direction to indicate direction of travel

lordfolken, 2023-04-26:
@MikeBravo15 could you design some aircraft icons, as you did for the waypoint icons?

Basically top down and clearly indicating a direction of travel?

The list would be the following:

[image]

lordfolken, 2023-04-26:
The idea is to display them on the map and the large traffic radar. I'll help you with the got stuff.

MikeBravo15, 2023-04-26:
I will try to draw something

MikeBravo15, 2023-05-02:
Do you want a dual set of icons? - ie "info" state and "warning" state (some red element)?

lordfolken, 2023-05-02:
I think we should color them at runtime. So please do them white or black on a transparent background.

MikeBravo15, 2023-05-02:
Work preview

[image]
Is white border OK?

Sundown3867, 2023-05-06:
Enclosed please find collection of icons 
[image]
[FlarmWarningIcons.zip](https://github.com/XCSoar/XCSoar/files/11412144/FlarmWarningIcons.zip)