XCSoar/XCSoar #922: Live Thermals from thermalmap.info not showing up

Developer66, 2022-05-03:
Live thermals from Thermalmap.info are not loaded into the map.

When activating the live thermals function, the thermals are not loaded into the map. Our server logs show that the data is picked up there.

Added Thermalmap's live thermals feature here: https://github.com/XCSoar/XCSoar/commit/aeacf63ad8ce6a091da470e8eb93575143663dc3

When I first looked through the source code, I couldn't find any reason why the thermals aren't appearing on the map. It would also be practical if the thermals appeared on the ground and not in flight, so that you could set a rough direction before your start.

Best regards
Tim Stuhler
www.thermalmap.info

XCSoar versions having and not having the problem
-------------------------------------------------
 > 7.22+

lordfolken, 2022-05-23:
What triggers an update?

Developer66, 2022-05-23:
I think it is somewhere called periodically in the ProcessTimer. But I am maybe wrong.
The data are periodically pulled every one minute.

https://github.com/XCSoar/XCSoar/blob/aeacf63ad8ce6a091da470e8eb93575143663dc3/src/ProcessTimer.cpp#L273

lordfolken, 2022-05-23:
I think this is missing some kind of logging or enduser notification.
Does this work in simulation mode?

closed 2022-08-24 by MaxKellermann