XCSoar/XCSoar #924: Flarm Declaration don't work with Flarm v7.06

August2111, 2022-05-06:
XCSoar versions having and not having the problem
-------------------------------------------------

From my point of view there has nothing to do with the XCSoar version, it seems to be a FLARM issue!

Expected behavior
-----------------

Declaration of task in Flarm should work
After sending of sentence 
```
$PFLAC,S,NEWTASK,Task*5B
```
the Flarm device should response with
```
$PFLAC,A,NEWTASK,Task*49
```
See FTD-014 from Flarm, and so the Flarm device with firmware v7.03 is doing!

Actual behavior
---------------
With Flarm v7.06 the answere is:
```
$PFLAC,A,ERROR*41
```
- and so you cannot send the WP to she device declaration...

After Sending 

Steps to reproduce the behavior
-------------------------------

Use Flarm v7.06

Do you have any idea what may have caused this?
-----------------------------------------------

FW issue

Do you have an idea how to solve the issue?
-------------------------------------------

Yes, I'm sending a bug report to the Flarm company - and expected a response. If there is some additional work to do as a workaround, I will do it - but I think there is a newer firmware from Flarm necessary (with solved issue on their side)!

August2111, 2022-05-06:
@lordfolken This seems to be a complete other issue then #906, where you announced this point:

SoaringMLJ, 2022-05-06:
Just to say that I have been able to declare to a classic FLARM unit with 7.06. The downloading is the only problem for me. 
Setup: XCSoar 7.23 on a HiSense A7CC hardwired via USB to a Soartronic IOIO device to classic FLARM 7.06.
I did have an earlier flight where I thought that I had declared but there was no declaration in the IGC file. But the FLARM was running 7.04 then and I updated to 7.06 as part of my troubleshooting. I was able to make a declaration and then read it back with the appropriate command, and a subsequent declaration made before a flight did appear in the IGC file.

August2111, 2022-05-06:
> Just to say that I have been able to declare to a classic FLARM unit with 7.06. The downloading is the only problem for me...

Flarm says the same to me, that they see also no problems ;-( But I know some guys, they couldn't declarate a new task - and nobody knows the difference... And I see the real issue during sniffering the device :( I want to do some more investigation in this issue - with a nearly contact to Flarm company itself!

SoaringMLJ, 2022-05-06:
Starting with the obvious - are they definitely IGC devices?

August2111, 2022-05-08:
Oops, I just have too many devices in the test - the 'error' only occurred with the NonIGC devices, big mistake on my part... :(
The bugs that were reported all only occurred with an OpenVario as host - maybe you have to look more there! I want to take a closer look at this next week!

closed 2022-05-08 by August2111