XCSoar/XCSoar #925: Flarm traffic not appearing on radar page

PNich, 2022-05-09:
On Kobo Aura 2nd ed running 7.23 flarm traffic appearing on map page pop up does not appear on flarm radar page.

From another user posted on hardware forum board:

As soon as you have a Traffic on a Map page ( Blue triangle), if you go in a Radar Page no Traffic reported, but if you press "info" 2/3 , and press
"traffic list", you can see the traffic, the callsign, distance, altitude, vario ecc. but, again No traffic reported on Radar page for many, many seconds.
XCSoar SW ver 7,23 with Phone Sansung and Powerflarm + ADSB
The same problem noticed on my OpenVario with the Stefly latest images.

lordfolken, 2022-05-12:
Do you have a way to reproduce this? I used the NMEA replay function on xscsoar, so everything should be as inflight, but cannot find a difference between the 3 pages, such as map/trafficgauge/flarmradar. 

Also any relation to #481

lordfolken, 2022-05-14:
So I finally found a way to reproduce this:
I asked some friends to borrow me their PowerFlarm Fusion. Thx (Salome&Alex!)

PowerFlarms have a simulation mode for Display Validation that can be triggered with ```$PFLAF,S,1``` (1 selects the scenario, there are 5)

Connecting to the powerflarm via tcp, and in pararllel connecting with 
nc <ip> <tcpport>
I can trigger the scenario 1 with ``$PFLAF,S,1```. This simulates a head on flarm target, starting far away and going through all stages.
Recording with the NMEA logger on in XCSoar settings, i cloud capture what the device outputs.

PNich, 2022-05-15:
That’s interesting – I wasn’t aware of that PowerFlarms had built in scenarios. Does this apply to the older PowerFlarms or just to the newer Fusion variants?

 

My next step was going to be to record the RS232 output from the flarm on a laptop app that can also play it back with reasonable time sync.  I have some RS232 files that I’ve used in the past in this way to try and debug a crashing problem on the club Kobos, but those files don’t have any embedded flarm traffic.  I could still make a recording if needed but I have some observations from various tests yesterday that I think might help pinpoint the problem:-    

 

This is the scenario using a Kobo Aura 2nd edition running 7.23 in my glider with lots of local flarm traffic around the airfield. Red box flarm providing GPS and Flarm data via serial to USB converter.

This set up all worked fine when running 6.8.11

 

1)      Initial power up

Pop up box on map screen appeared, but with no targets showing. “No Traffic” on Flarm radar page.  All traffic showing on traffic list page

2)      Enabled NMEA logging

Same behaviour as 1)

3)      Moved glider approx. 100m to launch point

Pop up box on map screen appeared but with targets showing this time. “No Traffic” on Flarm radar page. All traffic showing on traffic list page.

4)      Replayed NMEA file that had just been recorded

Pop up box on map screen + Flarm radar page + traffic list page all working correctly (same as you found)

5)      Multiple disabling /  enabling various flarm options in menus and recheck behaviour in normal “Fly” mode.

Same behaviour as 3)

6)      In flight

Pop up box on map screen + Flarm Radar page + traffic list all working correctly. (This surprised me as I believe that when I flew last week I thought the behaviour was as in 3), but I can’t be 100% sure)

7)      After landing 

Same behaviour as 3)

8)      Just now replayed NMEA file recorded during the flight yesterday.

Pop up box on map screen + Flarm radar page + traffic list page all working correctly (same as you found)

 

So, my conclusions:

 

·        Everything works when replaying an NMEA file

·        Correct data is getting to XCSoar 

·        The problem of not displaying targets correctly on pop up / flarm radar only occurs when receiving real time data from flarm and seems to be linked to the system knowing that it has moved since power up / is flying. 

I hope this helps

lordfolken, 2022-05-15:
> That’s interesting – I wasn’t aware of that PowerFlarms had built in scenarios. Does this apply to the older PowerFlarms or just to the newer Fusion variants? 

Since flarm finally has decided to actually publish their developer docs on their webpage, i can now link to them!
https://flarm.com/wp-content/uploads/man/FTD-012-Data-Port-Interface-Control-Document-ICD.pdf
Page 52, Section 8.18 PFLAF – Simulated traffic and alarms

Since its in the standard, all Flarms (including classic) should be able to trigger the simulation.

The issue here is really the use of the Replay function. Something gets treated differently when using that function, as compared to the refeeding the nmea log to xcsoar (I have attached the method to the bug report, please have a look)
https://github.com/XCSoar/XCSoar/issues/925#issuecomment-1126814189

Also what would help would be to check with more recent XCSoar versions. such as 7.10 etc. The gap between 6.8 and 7.24 is huge.

PNich, 2022-05-15:
OK, understood.

 

In that case I think the best way I can help is to capture “live” serial data from a flarm and check I can simulate the problem on the bench. I should have physical access to the data stream in one of the club gliders next weekend.  

 

If this works I’ll be able to check the behaviour of various XCSoar versions. What versions do you suggest I try as I only have access to Kobos and I know there were issues with the Kobo builds for the early 7.XX versions

 

https://flarm.com/wp-content/uploads/man/FTD-012-Data-Port-Interface-Control-Document-ICD.pdf

#925 (comment) <https://github.com/XCSoar/XCSoar/issues/925#issuecomment-1126814189>

lordfolken, 2022-05-15:
You can simulate the nmea feed with 

```bash 
make output/UNIX/bin/FeedMEA
FeedNMEA tcp 4345 < pflaf01.txt
``` 

No need for hardware. 
What is more interesting is to find out which version broke it.

PNich, 2022-05-15:
I assume this is a way to force PowerFlarm to output the built in scenario 1 via a tcp connection, but unfortunately I don’t have access to any spare flarm devices at the moment (and my S/W expertise is very limited).

 

I think I’ll have to stick to my H/W approach

lordfolken, 2022-05-16:
So already 7.5 seems to  have the problem.

lordfolken, 2022-05-16:
6.8.17 is the last version where the problem doesn't appear.

PNich, 2022-05-16:
So something in the major V7 uplift!

 

I still plan to do the hardware capture of Flarm NMEA output at the weekend – as a minimum it might be useful in testing.  Let me know if there is anything else I can do to help (with just some Kobos and Windows machines.)

lordfolken, 2022-05-16:
In your device configuration, is the FLARM device on device A or B?

lordfolken, 2022-05-16:
Ok so I found the commit that breaks it: 6ed282c5e158f9873ccdc9d5765fd3abf08e09e3

MaxKellermann, 2022-05-16:
See https://github.com/XCSoar/XCSoar/pull/63

PNich, 2022-05-21:
Happy to test – can someone provide a candidate build for Kobo please

 

Please test.

PNich, 2022-05-22:
I have recorded the NMEA stream coming out of a Red Box flarm during a couple of flights yesterday and these show the problem with 7.23. Unfortunately I don’t have the expertise or the equipment to compile a Kobo build but if anyone can provide a Kobo build I’ll test it.  I can provide the NMEA serial files if anyone wants them – they run on a simple windows app – “SerialPlayer Version 2.4”

closed 2022-05-22 by lordfolken

PNich, 2022-10-11:
Hi Philipp, I will do more tests at the weekend.  I think the problem in #925 is more basic than the enhancement in #481.

 

For a long time I had been running 6.8.11 on the club Kobos and everything worked well.  I updated them all to 7.23 (to get sounds which now works great with my external device  – thanks all) and now nothing ever appears on the flarm radar page I’d previously set up even though the targets appeared on the flarm pop up on the main map page.  

Has something changed in the config structure that has resulted in data heading for the flarm radar page no longer being enabled?

 

Also , for my understanding …what do you mean by “trafficgauge”?

 

Thanks

 

Pete

PNich, 2022-10-11:
It’s on device A and uses the FLARM driver

PR #930 (2022-05-22): Flarm Traffic is displayed again on the Traffic Gauges
fixes #925

The FLARM Traffic can be replayed with:
```bash
make output/UNIX/bin/FeedNMEA
./output/UNIX/bin/FeedNMEA tcp 4353 < ./test/data/driver/FLARM/pflaf01.nmea
```
And configuring a tcp client with driver flarm on 127.0.0.1 port 4353