XCSoar/XCSoar #933: replay +10' not working

fronzee, 2022-05-24:
XCSoar versions having and not having the problem
-------------------------------------------------
not sure

Expected behavior
-----------------

when opening replay and pressing +10' it should advance log file 10min ahead. 

Actual behavior
---------------

Does nothing. 

Steps to reproduce the behavior
-------------------------------

choose file to replay - press +10'

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2022-05-24:
Can you attach the igc file?

fronzee, 2022-05-24:
[2022-05-23-XCS-XYA-01.zip](https://github.com/XCSoar/XCSoar/files/8765851/2022-05-23-XCS-XYA-01.zip)
igc itself not allowed to upload, so compressed attached

lordfolken, 2022-05-24:
Seems to happen with any igc file i tried.

fronzee, 2022-05-25:
In addition to +10' I would say -10' would be good to have too, in case you go 10' too much in a 5hours flight... There would be a nice chance to just go back 10' instead of doing it all again from the beginning.

lordfolken, 2022-05-25:
Yeah the function needs work. There is also a demo mode if you leave the filename empty and press play. Very intuitive. 🙃

kobedegeest, 2026-01-22:
@coderabbitai have a look and propose a fix

hor63, 2026-10-16:
Not sure if I really want a true time jump. I kinda like the fast-forward playback for 10 min. This still shows everything.
I found the feature particularly useful at the start of the fight because the IGCs start a couple minutes before the take-off.

For a quick jump I would actually prefer an input window where you can

- enter a relative jump in min:sec like "+10:00" or "-05:05" (Yes! Jump back)
- absolute time into flight "03:20:32"

And leave the fast-forward as is.