XCSoar/XCSoar #937: Takeoff detection issue

christoph3761, 2022-06-04:
Hi,

in version 7.23 linux running on an OpenVario, a takeoff is detected immediately after starting XCSoar. It should be detected during the actual takeoff. I have two devices connected via the TTY ports of the OpenVario. This issue leads to wrong flight time calculations.

Christoph

lordfolken, 2022-06-05:
Hi Christoph,

The Takeoff detection of xcsoar is as follows:
https://github.com/XCSoar/XCSoar/blob/6f83f12b9243c0f80fc50410cef6b57d0846d5a6/src/Computer/FlyingComputer.cpp#L80-L90

So in order for XCSoar to detect a takeoff there has to be movement of the gps position. Perhaps a bad sensor?

christoph3761, 2022-06-05:
Hi Philipp,

thanks for following up on this! I am currently testing different versions of OpenVario software due to the issue mentioned in https://github.com/Openvario/meta-openvario/issues/304. My fallback variant is OpenVario build 17119 which uses an older beta version of XCSoar 7.0. With an unchanged hardware setup and XCSoar 7.0 beta instead of 7.23, there is no takeoff-detection after starting XCSoar. So I guess something has changed in the takeoff detection in between the XCSoar versions, at least for the OpenVario software.

Cheers
Christoph

lordfolken, 2022-06-05:
Well looking at the history there are about 3 commits that come into question:
https://github.com/XCSoar/XCSoar/commits/master/src/Computer/FlyingComputer.cpp
90e37270762eb838549075d677fc09dd00fd6dcf
57a2cf234247757c35ce2eb1ea74f5390b51c908
18b31a20b158758674a93c262a96654f25da51b8

lordfolken, 2022-06-05:
Can you turn on the nmea and auto logger (setup-logger in the configuration) and restart your device? Then please supply us with the nmea log. That way it can be replayed with FeedNMEA (make output/UNIX/bin/FeedNMEA) and we can reproduce.

christoph3761, 2022-06-05:
Will do. This may take a couple of days.

christoph3761, 2022-07-19:
Sorry for being unresponsive here, my glider was damaged during an outlanding and is still in a professional workshop. Will get the NMEA log once I have it back.

lordfolken, 2022-08-23:
Reopen when you can provide the info.

christoph3761, 2022-08-26:
I finally picked up the glider again last weekend and took 2 NMEA logs with XCSoar version 7.23

[2022-04-28_17-42.nmea.txt](https://github.com/XCSoar/XCSoar/files/9435475/2022-04-28_17-42.nmea.txt)
[2022-04-28_17-50.nmea.txt](https://github.com/XCSoar/XCSoar/files/9435476/2022-04-28_17-50.nmea.txt)

christoph3761, 2022-08-26:
@lordfolken seems I cannot reopen this issue. Do you want a new one or can you reopen it?

MaxKellermann, 2022-09-01:
I can't reproduce the problem by replaying those files. Can you, @christoph3761?
What is your displayed airspeed and AGL altitude when XCSoar detects takeoff?

christoph3761, 2022-09-01:
Hi @MaxKellermann, the displayed IAS is 27kph while on the ground in the hangar. This value is provided by the NMEA data of a Butterfly-Vario. I recently had the system status remotely checked by the manufacturer who stated that the total pressure sensor has an offset and needs to be recalibrated (the vario creates a system data dump for this purpose). This might cause the issue of XCSoar starting the flight although the glider is still standing on the ground. While have the unit checked by the manufacturer in winter.

I need to check on AGL altitude. I only remember that MSL altitude was slightly fluctuating by a few meters per second (perhaps +/- 3m/s) while standing on the ground. I am not sure which data stream is used by XCSoar. There are two devices connected providing barometric altitudes, the Butterfly Vario and a PowerFlarm unit. Barometric altitude readings on the ButterFly Vario are very constant.

closed 2022-09-01 by MaxKellermann