XCSoar/XCSoar #939: XCSoar boots, but nothing can be tapped on (kobo nia)

stefanoborini, 2022-06-05:
XCSoar versions having and not having the problem
-------------------------------------------------

I am running with 7.23-kobo, installed on a brand new Nia. The reader works fine with the basic system.

Actual behavior
-----------------

I installed XCSoar by dropping the kobo image under .kobo, and the program starts correctly, however I only get to the screen where it asks "Fly * Sim". Any tapping with the finger does not do any action. I see additional buttons "Nickel", "Tools", "Network", "System" and "Poweroff" but those cannot be clicked either.

Expected behavior
---------------

That it works, or gives some kind of message. Unsure if it's frozen, or it is working but not accepting any input.

lordfolken, 2022-06-05:
nia support was added here: 
https://github.com/XCSoar/XCSoar/pull/635
https://github.com/XCSoar/XCSoar/pull/634
@vkns can you assist?

stefanoborini, 2022-06-25:
@lordfolken if the support has been added, I guess it's just a matter of using a development version, am I correct?

lordfolken, 2022-06-25:
@stefanoborini yes. you can either build it on your own, or download from the github actions page.

stefanoborini, 2022-06-25:
@lordfolken unfortunately it does not seem to be fixed. I just installed git 8a11b88 (the current master) and I see the same issue.

lordfolken, 2022-06-25:
@vkns implemented this. Personally i dont have such a device.

lordfolken, 2023-02-26:
@XCSoar/kobo-team can someone of you test this? I lack such a device.

GeekGoneOld, 2023-02-26:
Some of the other models had similar problems.  They were "fixed" by putting a delay in rcS, currently line 87 just before loading kernel modules.  Recently a PR was proposed with this increased from 1s to 3s.  Maybe the 1s delay is just a bit too short to handle all cases (including newer Kobo f/w).  Perhaps increasing the delay in rcS would "fix" the NIA issue.

lordfolken, 2023-02-26:
You are talking about: 
https://github.com/XCSoar/XCSoar/commit/7bc9dd41ecc61ca16ea7e7c34d2f2666ab895f1c

@stefanoborini can you try a current build?

blackflag161, 2023-02-27:
I have a bit of a wild suggestion. As I recently got into flying, I got a Clara 2E after seeing it had been added to the development builds. So I loaded #91ea157 (sorry, can't find the link, that's what I got pulling with `gh run -R XCSoar/XCSoar download`). At first, it worked fine. After some playing around with it in Condor 2, I encountered the same issue as the OP, but I found something interesting you might try to reproduce:

@stefanoborini When restarting the Kobo, XCSoar seems to freeze/not accept input. BUT when I plug the device in  to USB power, it works fine again. When in XCSoar, if I unplug the power, it become slow and responds only to a fraction of inputs and generally tends to not let you switch pages/move and zoom the map. When I plug the USB-C back in, it works perfectly the moment it receives power.

If this is not the same bug, I can add a dedicated issue for the Clara 2E (refraining from this right now, as obviously I'm using a dev build, so I do know YMMV in that case).

SolidRick, 2024-04-28:
Hello to all,
it seems I'm having a similar problem. The newest Kobo Nia goes into power saving mode if "nothing" happens for 1-3min, then XCsoar is frozen and factory reset is the only cure. The problem is how to avoid such stall: based on preliminary tests, if I launch XCsoar and switch on the GPS just after (I have a switch on TX data of the GPS module to allow boot), Kobo Nia doesn't go into power saving and everything works fine. Now trying more cases

lordfolken, 2024-10-20:
I can no longer reproduce this with 7.44 on nia.

closed 2024-10-20 by lordfolken