XCSoar/XCSoar #948: Raspberry Pi4 XCSoar v7.24 crahes with floating point exception on startup

phintenaus, 2022-06-19:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2022-06-19:
Can you specifiy which git commit you took? and can you add more log lines?
Potentially can you start with 
```strace -fff -o mycrashlog.txt xcsoar```
and attach the log?

bomilkar, 2022-06-20:
Did you try to start xcsoar with explicit DPI argument?
`xcsoar -dpi=168x177`
Does it still produce the exception?

lordfolken, 2022-08-23:
No info since 2 months. closing.

closed 2022-08-23 by lordfolken