XCSoar/XCSoar #950: Airspace warnings not displayed on map screen when returning from flarm radar page

PNich, 2022-06-23:
XCSoar versions having and not having the problem
-------------------------------------------------
Found on 7.24 on Kobo Aura 2nd ed when testing fix to flarm radar screen not updating bug (that fix works OK).
Reproduced on 7.23 and 6.8.11

Expected behavior
-----------------
If when looking at Flarm radar page you get an audio airspace warning (near or in airspace) and you then return to map screen, you should be able to see warning message specifying what airspace had triggered the warning. 

Actual behavior
---------------
No warning message is displayed when returning to map screen after audio warning received when looking at Flarm radar page. 

Steps to reproduce the behavior
-------------------------------
Can be reproduced when replaying .nmea file having suitable airspace incursions.  

Do you have any idea what may have caused this?
-----------------------------------------------
No

Do you have an idea how to solve the issue?
-------------------------------------------

Ensure any active airspace warnings are displayed when returning to map screen.
Add airspace warning message display to Flarm radar page?

lordfolken, 2022-06-23:
Yeah the airspace warning should overdraw anything.

groundhogxc, 2022-08-17:
The exact same behavior also occurs when the alert happens in panning mode. When returning from panning to normal map display, the warning box does not show.

Scumi, 2023-06-14:
To add: You also do not see any visual warning in the Flarm radar screen itself (which I unfortunately discovered recently...). My strong suggestion would be to also display the warning there.

@lordfolken I like your suggestion in #200. This could solve both issues, no?

kobedegeest, 2026-08-30:
@coderabbitai investigate and propose a plan, maybe already take the changes of #2963 into account as well.

kobedegeest, 2026-08-30:
@coderabbitai warnings should also display when not on the map page. So forcing a re-check when entering map view alone is not enough

closed 2026-10-05 by lordfolken