XCSoar/XCSoar #951: IGC log files occassionally omit flight date record

dadyeah, 2022-06-23:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
----------------- Flight record date should be included but 30/12/1899 is found

Actual behavior
--------------- 30/12/1899 is found

Steps to reproduce the behavior
-------------------------------

 Not able but I have included the problem file.
The next flight on the same hardware worked correctly
My backup file on the day was another phone running XCSoar, it produced a correct file

Do you have any idea what may have caused this?.................... Maybe
-----------------------------------------------

 I had two GPS sources selected for redundancy (T-Beam and phone GPS)

Do you have an idea how to solve the issue?
-------------------------------------------  Ask Chris Fox to ask other pilots who have the same problem if they had two GPS sources selected

[2022-06-20-XCS-AAA-01_-_Copy[1].txt](https://github.com/XCSoar/XCSoar/files/8970319/2022-06-20-XCS-AAA-01_-_Copy.1.txt)

lordfolken, 2022-06-23:
Platform? Configuration?

lordfolken, 2022-06-23:
A nema logfile could also be of help.

lordfolken, 2022-06-23:
The record is only written after a time plausability check.
https://github.com/XCSoar/XCSoar/blob/464df291983948c67e967725e613b25c3efd9814/src/IGC/IGCWriter.cpp#L119-L123

dadyeah, 2022-06-24:
I pasted it into the box a the bottom but it must've failed, yer tisAndroid 9, T-Beam flarm and Motorola e6-plus GPS both selected.Chris Fox at the BGA Ladder says he's seen this before with XCSoar logs
Andy Smith

lordfolken, 2022-06-24:
XCSoar does not record the date, if it believes the date to be not realistic:
https://github.com/XCSoar/XCSoar/blob/8a11b887831a649e180b97f4e860590cd6a217aa/src/time/BrokenDate.hpp#L101-L105

Most likely one of your gps sources supplies a false or corrupted date.

tyddynonn, 2022-06-24:
This is Chris Fox - I run the BGA Ladder web site. 
IGC file parsing there will reject a file without a valid HFDTE record, and anecdotally I have seen this happen before with IGC files ex XCSoar. 
The snippets above from IGCWriter provides a plausible explanation and suggests the problem is with the date/time on the device.

lordfolken, 2022-06-24:
I'm closing this for now. If you can provide a NMEA log of this occurence, one could investigate. But really this has to be taken up with the manufacturer of these devices.

closed 2022-06-24 by lordfolken