XCSoar/XCSoar #952: unable to find information that allows me to progress with linux install on 5.4.0-121-generic #137-Ubuntu: $ sudo dpkg -i xcsoar_7.23_amd64.deb

nrbray, 2022-06-25:
XCSoar versions having and not having the problem
-------------------------------------------------
Which XCSoar version are you using when seeing the problem?
$ sudo dpkg -i xcsoar_7.23_amd64.deb

Which XCSoar version is the last one you know that did not show the problem?
<<7.23 installed in linux before...

Expected behavior
-----------------

Should install without error, or info in manual to install

Actual behavior
---------------

The following packages have unmet dependencies.
 xcsoar : Depends: libjpeg62-turbo (>= 1.3.1) but it is not installable
          Depends: liblua5.4-0 (>= 5.4.2) but it is not installable

Steps to reproduce the behavior
-------------------------------
sudo apt install -f ./xcsoar_7.23_amd64.deb 

gave the info message (actual behaviour)

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2022-06-25:
its a Debian package geared to Debian 11 (Bullseye).

maybe you can start engineering a process for the OpenSuSE Build Service. They build packages for all the distros. 
https://build.opensuse.org/

lordfolken, 2022-06-30:
I'm closing this as its currently not really supported.

closed 2022-06-30 by lordfolken