XCSoar/XCSoar #953: Reboot not working with latest KOBO Aura 2nd Ed device software update

PNich, 2022-06-26:
XCSoar versions having and not having the problem
-------------------------------------------------
6.8.11 , 7.23, 7.24 all show same problem following KOBO Aura 2nd Ed s/w update to latest version 4.33.129608 (5/31/22)

All XCSoar versions work OK with earlier KOBO s/w  4.15.12920 (5/27/19)

Expected behavior
-----------------
From front screen "System" then "Reboot" should cause device to Reboot
From front screen "System" then "Enable USB-OTG" should cause reboot with different driver.

Actual behavior
---------------
"Reboot" and "Enable USB-OTG" do not invoke reboot although USB-OTG is active following manual power cycling. "Disable USB-OTG" requires similar power cycling
"Nickel" and "Export USB storage" work as expected 

Steps to reproduce the behavior
-------------------------------
Press "System" then "Reboot"

Do you have any idea what may have caused this?
-----------------------------------------------
Change in Kobo device s/w (UI does look different!)

Do you have an idea how to solve the issue?
-------------------------------------------
No sorry

lordfolken, 2022-06-26:
@XCSoar/kobo-team anyone have that particular device and can investiage?

GeekGoneOld, 2022-06-27:
I have a flying buddy who uses a Aura 2nd Ed.  I will see if he is willing to update Kobo FW.

GeekGoneOld, 2022-06-27:
He has concerns that testing may break his Kobo.  I think I know what is failing and it is a trivial change.  In src/Kobo/System.cpp

  /* CLARA_HD requires the -f option to for reboot to work */
  if  (DetectKoboModel() == KoboModel::CLARA_HD)
  {
    return Run("/sbin/reboot", "-f");
  }
  return Run("/sbin/reboot");
#else
  return false;
#endif

add KoboModel::AURA2 to the "if" test

@PNich are you able to build a local copy or do you want me to build one and send the binary to you?

GeekGoneOld, 2022-06-27:
@PNich what flavour of Aura 2nd Ed do you have?  Specifically, is the 7th digit of your Kobo serial number a "2"?  Yes, it matters.

In the meantime, you can downgrade your Kobo F/W from https://pgaskin.net/KoboStuff/kobofirmware.html

GeekGoneOld, 2022-06-27:
@PNich you can test if my suggestion will work by telnetting into your kobo (e.g. over WiFi with PuTTY) and typing "/sbin/reboot" (without quotes).  If it doesn't reboot (shouldn't based on your experience), type "/sbin/reboot -f" and see if it does.  I'm hoping it does.  If so, I will gladly submit a PR.

PNich, 2022-06-29:
To answer the various questions:

 

·        I’m afraid I don’t have a build environment or the s/w expertise to build a local copy so I’ll need a .tgz file to test.

·        I have quite a few Kobo Aura’s as I put them in each of the club gliders as well as my own and a couple spare. They are all Kobo Aura Edition 2 v1 (the 7th digit is a 0 and not a 2). I also have an image of the original SD card for most of them so they can be recovered if they get bricked. I have some with latest Kobo firmware and some that have the much older version so I can compare easily. 

·        I’ll have a go at telnetting in and trying the different reset commands.

PNich, 2022-06-29:
I’ve just tried reboots via telnetting into 2 devices. Both running 7.24. Older Kobo firmware is 4.15.12920 (5/27/19) . Latest firmware is 4.33.129608 (5/31/22)

 

·        Typing /sbin/reboot with device with latest firmware does not force reboot

·        Typing /sbin/reboot with device with older firmware does force reboot

·        Typing /sbin/reboot –f  with device with latest firmware does force reboot

·        Typing /sbin/reboot –f  with device with older firmware does force reboot

 

So basically as expected J

 

The only thing I’ve noticed is the reboot timing is different with the –f option than without:

 

Without the –f option (on older firmware) the screen is blanked immediately the command is sent and then takes ~ 17 sec to get back to welcome screen

 

With the –f option (on both older and latest firmware) the screen is blanked ~ 6 sec after the command and it only takes a further ~ 4 sec to get to the welcome screen

 

It seems that the processes it’s executing are a bit different with the –f option, but as far as I can tell XCSoar works fine.

 

Happy to test a .tgz file if you can get me one.

lordfolken, 2022-06-29:
-f reboots directly without going through a shutdown procedure. The question is a bit why is it not shutting down normally.

GeekGoneOld, 2022-06-29:
Clara HD "fixed" this problem by using -f.  I agree with @lordfolken that the heart of the problem is why is Clara HD and Aura 2nd Ed (after a certain fw) not shutting down normally.

@lordfolken Should I submit a PR with -f on Aura 2nd Ed as well?  Although I see the weakness in that the "problem" hasn't been found or solved, it has served me very well on the Clara HD.

PNich, 2022-06-30:
In case it’s any use to you I’ve found that the most recent fw for the Kobo Aura 2nd Ed V1 that works ok is 4.21.15015 – May 2020. Anything newer than that has this problem.

lordfolken, 2022-06-30:
here we replace the initiab, which is responsible for startup shutdown reboot handling. 
https://github.com/XCSoar/XCSoar/blob/464df291983948c67e967725e613b25c3efd9814/kobo/inittab

orengt, 2022-11-01:
Hi all,

I run xc 7.27 on KOBO Aura 2 Ed. 
I met this problem in which kobo doesn't boot after enabling/disabling OTG as described here.
I fortunately I didn't understand the conversation above, can some one explain to a dummy like me if the problem should be  solve/will be solved/or will never be solved?

Thanks,
Oren

GeekGoneOld, 2023-02-08:
> Well i'm fine if this is a platform bug. I.e. has nothing to do with our modifications. If this is an issue caused by what we copy onto the root, then we have to fix it. Otherwise we risk bricking other applications / update processes etc of kobo itself.

I FINALLY figured out why you need the "-f" for certain models.  What a pain to track that one down!!!  Turns out newer versions of busybox (somewhere around 1.30?) won't reboot until init completes.  The last thing init does in XCSoar is to calls /opt/xcsoar/bin/rcS and the last line (normally) of rcS is "exec /opt/xcsoar/bin/KoboMenu" which means init never completes, it just runs KoboMenu and never returns.

PowerOff works because it is executed (actually initiated) by KoboMenu and after initiating poweroff, KoboMenu exits which exits init and allows the poweroff to complete.  With Reboot, though, it is initiated by SystemMenu (whether because of the reboot button or the OTG enable/disable) and after being initiated, it simply returns to the menu.

My proposed solution which seems to work perfectly on Clara HD (used reboot command from telnet to test) is to alter the last few lines of rcS from
```
# finally, open the menu
exec /opt/xcsoar/bin/KoboMenu

# just in case launching KoboMenu fails: try xcsoar
exec /opt/xcsoar/bin/xcsoar
```
to
```
# finally, open the menu
/opt/xcsoar/bin/KoboMenu &
```
Note that this does not maintain the "just in case" launch of XCSoar.  I am uncertain as to how to easily verify that KoboMenu worked so as to avoid the direct open of XCSoar and I'm not sure of its value anyway (could be VERY outdated).  If it is still necessary, I'm sure there is an easy to check if KoboMenu worked and I'm open to suggestions.

@lordfolken, @MaxKellermann , could you please use your experience to validate my repair (i.e. look for "gotchas") and if you find it acceptable I will write a PR (which will also remove the conditional -f on reboot).  If you have a problem with it, I still believe I have successfully tracked down the cause so you could offer some guidance as to how you would like it done (eg. causing reboot to exit out of SystemMenu and KoboMenu) and I can create a PR based on your critique.

Awaiting your guidance.

MaxKellermann, 2023-02-08:
> init completes

What does that mean, exactly?

GeekGoneOld, 2023-02-08:
> > init completes
> 
> What does that mean, exactly?

Simply, pid 1 (init) exits and pid 1 is no longer running so busybox can see that pid1 (init) is not running and assumes completed.  When we use exec, KoboMenu "takes over" pid 1 (init) and busybox reboot sees that init never completed.

MaxKellermann, 2023-02-08:
> pid 1 (init) exits and pid 1 is no longer running

But that would be wrong. pid1 must never exit, therefore cannot ever "complete" that way.

GeekGoneOld, 2023-02-08:
> But that would be wrong. pid1 must never exit, therefore cannot ever "complete" that way.

I just looked at ps and you are absolutely right, pid 1 (init) is still running.  I suppose busybox checking to see if init "completes" means that at a minimum it must return from rcS.

The solution I have proposed, therefore, allows init to reach sufficient completion (e.g. returns from rcS) to allow busybox to reboot) but does not allow init to exit.

MaxKellermann, 2023-02-08:
The proper way would be to add KoboMenu to inittab as a service, but for the Kobo that would probably be overkill; so let's `&` it as you suggested.

GeekGoneOld, 2023-02-08:
> The proper way would be to add KoboMenu to inittab as a service, but for the Kobo that would probably be overkill; so let's `&` it as you suggested.

Will do.  I will create a PR tomorrow that uses the & instead of exec and removed the direct launch of XCSoar if KoboMenu fails.

Once again, many, many thanks to you, Max.

commit d474753c (2023-02-17): Removed force on Kobo reboot

Newer busybox firmware requires rcS to finish, returning to init
in order for the reboot command to work.  Ending rcS with
exec /opt/xcsoar/bin/KoboMenu prevents this so reboot does nothing.
Creating a new process by removing the exec and running with no wait
allows reboot to work and is preferred for old or new Kobo f/w.
Closes #953
PR #1111 (2023-02-17): Removed force on Kobo reboot
Newer busybox firmware required changes to avoid using -f (force) when using the reboot command.
Closes #953

Brief summary of the changes
----------------------------

Related issues and discussions
------------------------------
closed 2023-02-17 by MaxKellermann