XCSoar/XCSoar #957: flight times rounding error

hjr, 2022-07-02:
XCSoar versions having and not having the problem
-------------------------------------------------

7.21 and older, most probably also 7.23.

Expected behavior
-----------------

Expect the page Status->Times to show flight time as difference of departure time and landing time. 

Actual behavior
---------------

The flight time is actually by 50/50 change different to that by one minute.
[image]

Steps to reproduce the behavior
-------------------------------

Fly, land, look on the status page.

Do you have any idea what may have caused this?
-----------------------------------------------

I guess difference, departure time and landing time are rounded to minutes independently, instead of doing the difference on already rounded departure and landing time.

Do you have an idea how to solve the issue?
-------------------------------------------

Round first the departure and landing time, then do the subtraction.

lordfolken, 2022-08-23:
Thx for picking this up.

MaxKellermann, 2022-08-23:
This means pretending to be more accurate while being less accurate, while increasing code complexity.

MaxKellermann, 2022-08-31:
If you launch at 10:02:00 and land at 10:07:59, how long was your flight?
- accurate value is 5m59s
- accurate truncated value would be 5 minutes
- accurate rounded value would be 6 minutes
- currently, XCSoar displays 5 minutes (= accurate truncated)
- after #1027 XCSoar would display 5 minutes

If you launch at 10:02:59 and land at 10:08:00, how long was your flight?
- accurate value is 5m1s
- accurate truncated value would be 5 minutes
- accurate rounded value would be 5 minutes
- currently, XCSoar displays 5 minutes (= accurate truncated)
- after #1027 XCSoar would display 6 minutes (= ???; *so* wrong, no matter how you look at it)

(Note: currently, XCSoar truncates both the start/landing time and the flight duration; no rounding. It can be debated whether truncation or rounding would be better, and if somebody feels XCSoar should round displayed values, I'd agree to change that.)

#1027 adds a second layer of truncation to calculate the flight duration, it truncates twice, losing more precision. This makes it less accurate than the current behavior, which truncates only once.

I think this whole thing is a non-issue that does not need to be addressed at all. This is just an artifact of displaying less precision on the UI than XCSoar has internally. Attempts to "solve" this non-issue with #1027 only lead to complicated code, bloat, reduced performance and fragile code.

hjr, 2022-09-01:
Ok, Max, I guess you did not understand the issue I described first hand. I do not care about rounding  or truncating, not at all. I just like to see a consistency on the GUI that shown flight time equals landing time - launch time. And that is not given. 

Picking up your second example .. nomatter what the flight time is more or less wrong. But the GUI will try to make you believe that the flight went from 10:02 to 10:08 and took 5 minutes. 
I just would like to see a match here that I can take over in my flight book. Currently it is a 50:50 chance of having an inconsistency then.
But good that we figured that the code change is not the issue, it is the issue itself. Here we are. Still a lot of taste and prospectives to align anyways.

hjr, 2023-07-09:
Hey lordfolken, what to do here? You put this issue on the to do list, but after this exchange of thoughts it is unclear to me how to procede.
Do we want to see a consistent GUI where start time and flight time adds up to a landing time, or do we not care about this inconsistency, prefering an "accuracy" on rounded time stamps, that leads to a not consistent flight time here.