XCSoar/XCSoar #958: Waypoint list displays duplicate entries for .cup

GeekGoneOld, 2022-07-03:
XCSoar versions having and not having the problem
-------------------------------------------------

7.18 is OK
7.19 has the problem
Possibly introduced in 9533b1221259257112d658a97474dddf9957cc03

Expected behavior
-----------------

When using guesture "L", I expect to see the consolidated list of waypoints with each one appearing only once.

Actual behavior
---------------

If a .cup waypoint has both a Title and Code (fields 1 and 2) non-blank, it appears in the waypoint list twice

Steps to reproduce the behavior
-------------------------------

Using any .cup file with a line that has both Title and Code (field 1 and 2), on the map display guesture "L" to call up the waypoint list and you will see that entry twice in the list (not necessarily consecutive).

Do you have any idea what may have caused this?
-----------------------------------------------

Perhaps it was a small error when the second field support was introduced in 7.19 or it has to do with filtering on both fields.  If it is the latter, I would hope a solution can be found to support filtering and only show once.

Do you have an idea how to solve the issue?
-------------------------------------------

Not really.  I'm unfamiliar with the code.  I'm happy to help if I can get guidance.

lordfolken, 2022-07-03:
duplicate of #816?

lordfolken, 2022-07-05:
There was a PR a while back: https://github.com/XCSoar/XCSoar/pull/743 but unfortunately its implementation didn't scale. Maybe you have a better idea?

GeekGoneOld, 2022-07-05:
Yup.  Got a better idea.  I’m working on NOT adding waypoint short names to the WaypointNameTree (Waypoints.cpp) then in WaypointFilter.cpp I will check if the Name filter matches either the name or the short name.  1 week tops...

lordfolken, 2022-07-05:
I'm very grateful for your efforts! :+1:

GeekGoneOld, 2022-07-05:
> I'm very grateful for your efforts! 👍

As I am for all of yours!!!

yo7ckp, 2022-07-05:
I gave up "Code" at waypoints.

kicune, 2022-08-14:
> Yup. Got a better idea. I’m working on NOT adding waypoint short names to the WaypointNameTree (Waypoints.cpp) then in WaypointFilter.cpp I will check if the Name filter matches either the name or the short name. 1 week tops...

If the short names won't be in the name tree, then SuggestNamePrefix will not consider them, therefore it will not be possible to enter the short name via the preemptive keyboard, no?

GeekGoneOld, 2022-08-15:
> If the short names won't be in the name tree, then SuggestNamePrefix will not consider them, therefore it will not be possible to enter the short name via the preemptive keyboard, no?

@kicune, you are correct.  It is actually very desirable to have the short names in the name filter so eliminating short names from the waypoint list without other consideration would be too limiting.

I found that just eliminating the names from the name tree turned out to bloat the code (needed to add two other name trees - one for shortname and one for both names together) and it was a mess.

I'm now looking at a rather trivial solution of sorting then "unique"ing.  @xtophe80 did something similar in PR#743 but his solution scaled with O(Nsquared).  A sort then unique solution will scale with O(NlogN), improving the performance.  Note that if the waypoint list is by distance, it is sorted before returning and that was considered acceptable for performance so I'm hoping @MaxKellermann will find it acceptable for names as well!

I won't promise a date for the PR as it is prime season!  I'm definitely hoping it is soon enough for review and revision before 7.25.

kicune, 2022-08-15:
I was thinking along the lines of PR#743, but with using stl::map<unsigned, WaypointPtr> with Waypoint.id as the key instead trivial sequence search. This should be O(n.log(n)) in the typical case.
But to be honest, with the realistic number of matches in the filter, I personally consider even O(n^2) as quite acceptable in this scenario (definitely more acceptable than number of duplicates that currently plaques my waypoint list). But that's what forks are for - I fully respect @MaxKellermann opinion.

commit 98a76d31 (2022-08-22): Eliminate duplicate waypoints in list

Retains filtering name and shortname but deletes duplicates
(e.g. when the shortname is the same as the name).  Also
eliminates duplicates when distance is the filter.  Bonus: this
should slightly speed up rendering of the waypoints on the map.
Closes #958
Closes #816
PR #983 (2022-08-22): Eliminate duplicate waypoints in list
Retains filtering name and shortname but deletes duplicates
(e.g. when the shortname is the same as the name).  Also
eliminates duplicates when distance is the filter.  Bonus: this
should slightly speed up rendering of the waypoints on the map.
Closes #958
Closes #816

Brief summary of the changes
----------------------------

Related issues and discussions
------------------------------
closed 2022-08-22 by MaxKellermann