XCSoar/XCSoar #962: xcsoar 7.23 crashes under Iinux / Ubuntu

phintenaus, 2022-07-05:
7.0 worked, 7.23 produces strange spikes when rendering the water area layer in the attached map before it crashes 

XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

/home/phintena/AlpsPlus.xcm

lordfolken, 2022-07-05:
The attachment is missing.
Please note: i just replaced the water areas in the lakes 2 days ago with more detailed versions. Please try those for your area.

MaxKellermann, 2022-08-19:
Crash confirmed, probably due to a triangulation bug.

MaxKellermann, 2022-08-20:
I think this is a Mesa / graphics driver bug.
I need your xcsoar.log to verify.

MaxKellermann, 2022-08-20:
For reference, my xcsoar.log:
```
GLX config: RGB=8/8/8 alpha=8 depth=24 stencil=8
GL vendor: AMD
GL version: 4.6 (Compatibility Profile) Mesa 22.2.0-rc2
GL renderer: AMD Radeon RX 480 Graphics (polaris10, LLVM 14.0.6, DRM 3.46, 5.18.12-heron)
GL extensions: GL_ARB_multisample GL_EXT_abgr GL_EXT_bgra GL_EXT_blend_color GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_copy_texture GL_EXT_subtexture GL_EXT_texture_object GL_EXT_vertex_array GL_EXT_compiled_vertex_array GL_EXT_texture GL_EXT_texture3D GL_IBM_rasterpos_clip GL_ARB_point_parameters GL_EXT_draw_range_elements GL_EXT_packed_pixels GL_EXT_point_parameters GL_EXT_rescale_normal GL_EXT_separate_specular_color GL_EXT_texture_edge_clamp GL_SGIS_generate_mipmap GL_SGIS_texture_border_clamp GL_SGIS_texture_edge_clamp GL_SGIS_texture_lod GL_ARB_framebuffer_sRGB GL_ARB_multitexture GL_EXT_framebuffer_sRGB GL_IBM_multimode_draw_arrays GL_IBM_texture_mirrored_repeat GL_ARB_texture_cube_map GL_ARB_texture_env_add GL_ARB_transpose_matrix GL_EXT_blend_func_separate GL_EXT_fog_coord GL_EXT_multi_draw_arrays GL_EXT_secondary_color GL_EXT_texture_env_add GL_EXT_texture_filter_anisotropic GL_EXT_texture_lod_bias GL_INGR_blend_func_separate GL_NV_blend_square GL_NV_light_max_exponent GL_NV_texgen_reflection GL_NV_texture_env_combine4 GL_S3_s3tc GL_SUN_multi_draw_arrays GL_ARB_texture_border_clamp GL_ARB_texture_compression GL_EXT_framebuffer_object GL_EXT_texture_compression_s3tc GL_EXT_texture_env_combine GL_EXT_texture_env_dot3 GL_MESA_window_pos GL_NV_packed_depth_stencil GL_NV_texture_rectangle GL_ARB_depth_texture GL_ARB_occlusion_query GL_ARB_shadow GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_mirrored_repeat GL_ARB_window_pos GL_ATI_fragment_shader GL_EXT_stencil_two_side GL_EXT_texture_cube_map GL_NV_copy_depth_to_color GL_NV_depth_clamp GL_NV_fog_distance GL_NV_half_float GL_APPLE_packed_pixels GL_ARB_draw_buffers GL_ARB_fragment_program GL_ARB_fragment_shader GL_ARB_shader_objects GL_ARB_vertex_program GL_ARB_vertex_shader GL_ATI_draw_buffers GL_ATI_texture_env_combine3 GL_ATI_texture_float GL_EXT_depth_bounds_test GL_EXT_shadow_funcs GL_EXT_stencil_wrap GL_MESA_pack_invert GL_NV_primitive_restart GL_ARB_depth_clamp GL_ARB_fragment_program_shadow GL_ARB_half_float_pixel GL_ARB_occlusion_query2 GL_ARB_point_sprite GL_ARB_shading_language_100 GL_ARB_sync GL_ARB_texture_non_power_of_two GL_ARB_vertex_buffer_object GL_ATI_blend_equation_separate GL_EXT_blend_equation_separate GL_OES_read_format GL_ARB_color_buffer_float GL_ARB_pixel_buffer_object GL_ARB_texture_compression_rgtc GL_ARB_texture_float GL_ARB_texture_rectangle GL_ATI_texture_compression_3dc GL_EXT_packed_float GL_EXT_pixel_buffer_object GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_mirror_clamp GL_EXT_texture_rectangle GL_EXT_texture_sRGB GL_EXT_texture_shared_exponent GL_ARB_framebuffer_object GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_EXT_packed_depth_stencil GL_ARB_vertex_array_object GL_ATI_separate_stencil GL_ATI_texture_mirror_once GL_EXT_draw_buffers2 GL_EXT_draw_instanced GL_EXT_gpu_program_parameters GL_EXT_gpu_shader4 GL_EXT_texture_array GL_EXT_texture_compression_latc GL_EXT_texture_integer GL_EXT_texture_sRGB_decode GL_EXT_timer_query GL_OES_EGL_image GL_AMD_performance_monitor GL_EXT_texture_buffer_object GL_AMD_texture_texture4 GL_ARB_copy_buffer GL_ARB_depth_buffer_float GL_ARB_draw_instanced GL_ARB_half_float_vertex GL_ARB_instanced_arrays GL_ARB_map_buffer_range GL_ARB_texture_buffer_object GL_ARB_texture_rg GL_ARB_texture_swizzle GL_ARB_vertex_array_bgra GL_EXT_texture_swizzle GL_EXT_vertex_array_bgra GL_NV_conditional_render GL_AMD_conservative_depth GL_AMD_depth_clamp_separate GL_AMD_draw_buffers_blend GL_AMD_seamless_cubemap_per_texture GL_AMD_shader_stencil_export GL_ARB_ES2_compatibility GL_ARB_blend_func_extended GL_ARB_compatibility GL_ARB_debug_output GL_ARB_draw_buffers_blend GL_ARB_draw_elements_base_vertex GL_ARB_explicit_attrib_location GL_ARB_fragment_coord_conventions GL_ARB_provoking_vertex GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_seamless_cube_map GL_ARB_shader_stencil_export GL_ARB_shader_texture_lod GL_ARB_tessellation_shader GL_ARB_texture_buffer_object_rgb32 GL_ARB_texture_cube_map_array GL_ARB_texture_g
Display dpi=96,96
OpenGL: mda=1 npot=0 stencil=0
```
Crashes here, triggered by `glMultiDrawElementsEXT()` call:
```
==2435253==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000006 (pc 0x7f6c82ac3b68 bp 0x631000050930 sp 0x7f6c75cb8800 T5)
==2435253==The signal is caused by a READ memory access.
==2435253==Hint: address points to the zero page.
    #0 0x7f6c82ac3b68 in amdgpu_cs_add_buffer ../src/gallium/winsys/amdgpu/drm/amdgpu_cs.c:674
    #1 0x7f6c82d8f235 in radeon_add_to_buffer_list ../src/gallium/drivers/radeonsi/si_pipe.h:1974
    #2 0x7f6c82d8f235 in si_emit_draw_packets<(amd_gfx_level)10, (si_has_ngg)0, (si_is_draw_vertex_state)0> ../src/gallium/drivers/radeonsi/si_state_draw.cpp:1533
    #3 0x7f6c82d8f235 in si_draw<(amd_gfx_level)10, (si_has_tess)0, (si_has_gs)0, (si_has_ngg)0, (si_is_draw_vertex_state)0, (util_popcnt)0> ../src/gallium/drivers/radeonsi/si_state_draw.cpp:2582
    #4 0x7f6c82d8f235 in si_draw_vbo<(amd_gfx_level)10, (si_has_tess)0, (si_has_gs)0, (si_has_ngg)0> ../src/gallium/drivers/radeonsi/si_state_draw.cpp:2634
[... cut ...]
==2433851==    by 0x7B2C5EE: _mesa_DrawElements (draw.c:1955)
==2433851==    by 0x1CA9BD: TopographyFileRenderer::Paint(Canvas&, WindowProjection const&) (TopographyFileRenderer.cpp:293)
==2433851==    by 0x1CBF1A: TopographyRenderer::Draw(Canvas&, WindowProjection const&) (TopographyRenderer.cpp:45)
==2433851==    by 0x25F5BA: Draw (CachedTopographyRenderer.hpp:56)
==2433851==    by 0x25F5BA: MapWindow::RenderTopography(Canvas&) (MapWindowRender.cpp:95)
```
(This is Mesa 22.2.0~rc2-1 from Debian Bookworm)

closed 2022-08-22 by MaxKellermann