XCSoar/XCSoar #966: FT232 Dual Channel Support

XCNav, 2022-07-08:
XCSoar version
--------------

7.2.2

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

The D2xx driver allows to distinguish between 2 devices with same vid/pid via openByindex. 

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
connectinc 2 serial devices in parallel operation on designated android device
parallel input of 2 serial datastreams

lordfolken, 2022-07-08:
This should be a bug. As this is unpredictable. Platform is android i assume?

lordfolken, 2022-07-08:
Also what version are you using? Can you add screenshots? And xcsoar.log?

lyusupov, 2022-07-10:
@XCNav XCSoar had inherited "_USB for Android_" support from LK8000 last year.
Thus, just like the LK8000 does, the XCSoar uses https://github.com/mik3y/usb-serial-for-android component to operate with USB-Serial devices.
The library is maintained by Kai Morich.. There is a reference application for Android that uses this library: - https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal - it was created by the same author.

You can test first if the reference app can handle multiple connections to FT232 devices at the same time. If it doesn't - it reduces the chances of having this feature in XCSoar.

lyusupov, 2022-07-10:
Inspect if your issue report duplicates this one: https://github.com/mik3y/usb-serial-for-android/issues/428

XCNav, 2022-07-10:
No, it does not exactly, as I have acces to both ports simultaneously. 
I am no software guy and dont work on a console and at the present I have no access to the device and will add Screenshots later. I can run both ft232 in parallel on the device using the usb serial monitor as above plus another serial monitor in splitscreen mode, both running different varios(xcvario and LXEra), both sending NMEA datastream.
I had the same issure running LK8000   
After communicating with Bruno from LK8000 he changed the way the driver identifies the 232. Now I can run both ports in parallel on LK8000.

lyusupov, 2022-07-11:
> After communicating with Bruno from LK8000 he changed the way the driver identifies the 232.

If would supply commit ID of the LK8000 patch - it will ease for XCSoar developers to track and resolve the issue.

brunotl, 2022-07-12:
LK8000 patch : commit  [d8cdbbd](https://github.com/LK8000/LK8000/commit/d8cdbbde75fe543957e7a2377b473890a35a0875)

XCNav, 2022-08-04:
As brunotl submitted the patch, is there any plan or progress?

hjr, 2022-08-18:
Have the same issue with XCSoar 7.23. Described it in a comment to #968

XCNav, 2022-08-23:
Is there more information needed than the patch? Should i open it as pull request?

lordfolken, 2022-08-23:
Please make a PR @XCNav will discuss in there.

MaxKellermann, 2022-08-24:
I thought the LK8000 code isn't good, so I implemented my own version of the feature, after refactoring the hell out of class UsbSerialHelper. It's now much more robust than before, and supports multiple ports per device, and multiple devices of the same kind.

closed 2022-08-24 by MaxKellermann

XCNav, 2022-08-24:
Perfect! Thank you!

XCNav, 2022-08-24:
[image]
Works perfect! Thank you Max!

hjr, 2022-08-26:
[image]
[image]
Unfortunately it does not work for those CP2102 serial bridges. I get the connect popup when I plug each of them, but the device enumeration shows only one. Selecting it a second time at least shows the consequent "already occupied" message.

lyusupov, 2022-08-26:
```
$ lsusb -v -d 10c4:ea60

Bus 001 Device 053: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x10c4 Cygnal Integrated Products, Inc.
  idProduct          0xea60 CP210x UART Bridge / myAVR mySmartUSB light
  bcdDevice            1.00
  iManufacturer           1 Silicon Labs
  iProduct                2 CP2102 USB to UART Bridge Controller
  iSerial                 3 0001
  bNumConfigurations      1
...

```

'Serial Number' (**iSerial** field above) of the CP210x device has to be a unique identifier across all other CP210X devices you plug into the XCSoar Android box.

One may alter the serial number by this utility: https://github.com/VCTLabs/cp210x-program