XCSoar/XCSoar #968: FTDI4232 (4-port USB to RS232) -- only the first port available on Android platform

henrik1g, 2022-07-12:
Chip datasheet:
https://ftdichip.com/products/ft4232hq/ 

XCSoar versions having and not having the problem
-------------------------------------------------
Linux is not affected since it natively identifies the FT4232 ports as ttyUSB[0-3]
Android Version 7.23 is affected

Expected behavior
-----------------
In the device/port dialog all four channels of the FT4232 should show up as separate ports so each can be assigned to an individual driver, so that multiple RS232 devices can be connected.

Actual behavior
---------------
In the device/port dialog only the first channel of the FT4232 is available (labeled "Quad RS232-HS(FTDI)" ).
This port works as expected. But the remaining three ports do not show up and therefore are unavailable.

Steps to reproduce the behavior
-------------------------------
Connect FTDI4232 to USB(-OTG) of an android phone.
I am using this board:
https://www.aliexpress.com/item/1005001928940997.html 

I am of course available for testing new code.

Side note:
Using this: USB-C-(OTG)-PD adapter 
https://de.aliexpress.com/item/1005003887696758.html
the (PD-enabled) phone can get reliably charged while acting as USB-host for the FT4232, so this is a viable setup and will be interesting for many people.
(The phone needs to be capable of USB-C-PD (PowerDeliveryProtocol) though)
Tested on Pixel4a.

Do you have any idea what may have caused this?
-----------------------------------------------
driver implementation of this chip in XCSoar does not account for multiple ports

Do you have an idea how to solve the issue?
-------------------------------------------
update driver code to assign all available ports
might be resolved together with 
https://github.com/XCSoar/XCSoar/issues/966 (not a duplicate though)

henrik1g, 2022-07-12:
When using a serial terminal app on the android phone. All four ports show up and are functional.
Latest version of LK8000 has the same problem as XCSoar though.

hjr, 2022-07-12:
Did you double-check on this is not an issue of the Android you are using? Could you confirm that the OS does recognize all FTDI ports?

henrik1g, 2022-07-12:
Yes. All four ports are working on the same phone. Using a serial terminal app.

henrik1g, 2022-07-21:
Looking at the code it seems to me that the solution needs to start here:
https://github.com/XCSoar/XCSoar/blob/2dca7496a558f0c963631ac79e56352a72d0d220/android/src/UsbSerialPort.java#L62
The call to the overloaded function createUsbSerialDevice() omits the option for the index of the interface, which makes it default to interface 0.

See here for the function including the interface number:
https://github.com/felHR85/UsbSerial/blob/7fff8b6d5ca19590dcb05c3f977970e8cce103b7/usbserial/src/main/java/com/felhr/usbserial/UsbSerialDevice.java#L66

hjr, 2022-08-16:
Hey Henrik,
this seems to be an easy issue to solve. I would be also interresting in your solution. Have you finished the hardware and have pieces available?
I tried to just connect multiple dongles through a regular USB hub. I experienced that devices with identical Manufacturer Id and Product Id do only show up once on my Android Reader. This sounde weired and not correct reading through https://microchipdeveloper.com/usb:enumeration I would expect that every detected USB device will get a unique address assigned to communicate to.

henrik1g, 2022-08-16:
Since up to now noone of the experienced developers showed interest in this and stepped forward to help and I happen to have some quarantine time right now, I am trying to wrap my head around the code in this particular area to come up with a pull request myself that might be eventually accepted.
That to say my programming experience on such a big piece of software is very very limited. So I find it hard to understand how excatly the broadcasting and such works within XCSoar and how the connection between the C++ world and the Android Java World is working.
In what I succeeded so far:
- The thirdparty tool on which our UsbSerial support is based:
https://github.com/XCSoar/UsbSerial/tree/461648295bb5937505d64d7fbead7a7e4632ba34
has full support for the FTDI4232
- the implementation in (android/src/) UsbSerialPort.java and UsbSerialHelper.java however is flawed in such a way that it does not honour multiple interface of a usb device. Instead it uses only the first one.
- I can successfully use each of the four interfaces of the FTDI by hardcoding the interface_number (one at a time). That is easy.
- What is left to do is find a way to provide a way to manage all interfaces simultaneously. Which (for me!) is harder than I originally thought. Since UsbSerialHelper has no provisions for additional interfaces and assumes a one to one relationship of Device and Port.
- I'm sure this would not be a big deal for someone familiar with that part of the code. However for me it is a challenge, but I'm willing to take it.

Any help is highly appreciated! =)

henrik1g, 2022-08-16:
Maybe your problem with similar adapters can be adressed in the same go.
It's both related to how the Devices (and soon better "Interfaces") are labelled and kept track of by UsbSerialHelper.java

henrik1g, 2022-08-17:
I have some more success to report. =)
With some modifications to UsbSerialHelper.java I am now able to access all four Interfaces via the Port picker menu and use them.
While changing the code I also wanted to address Helmuts problem of using several identical USB-Serial adapters. However thinking about it this might introduce another problem, and I would like some input of the devs that will eventually have to approve the change:

The chosen Android port is identified from the main app by:
 the type (=UsbSerial), 
a String containing the  "address" or id, 
(and a name and feature flags)

Up to now UsbSerialHelper uses a string containing VID:PID as id. 
To handle devices with more than one interface, or multiple devices with similar VID:PID this is not sufficient as "id"
In my momentary solution I created for each interface an id-string containing the usb-bus address extended by the interface(=port) number. (e.g. "/dev/bus/usb/001/002P0")
As stated the solution works fine, and should solve multiple interface and multiple similar VID:PID problems.
However it might introduce a problem if the bus allocation changes or is not always the same. Then a saved port would have to be reselected in the settings, while with the VID:PID string solution, the device should work just fine.
The multiple interface issue could be solves nevertheless just by making the id-string "VID:PID:Interface" (like "4003:6011:0") this would keep saved connections valid even if the location on the usb-bus changes. But it would not solve Helmuts problem.

Atm I prefer the second approach. ("4003:6011:0") over ( "/dev/bus/usb/001/002P0"). Or is there maybe a real unique identifier of a usbdevice I could use?
What is your opinion? Which way should I go to get the patch approved?

Cheers Henrik

henrik1g, 2022-08-17:
Tested working with FTDI4232 using the VID:PID:Interface scheme:
https://github.com/henrik1g/XCSoar/tree/UsbSerial_MultiInterface

hjr, 2022-08-18:
I like the VID:PID:Interface  scheme, because it give you as xcsoar user the oportunity to see which hardware the device is dealing with. Otherwise I think they are identically.
Yes, the niceness of reconnecting after a reboot to the same devices behind the interface is not granted. I would just hope that Android will always enumerate the same way.

henrik1g, 2022-08-18:
I also prefer the VID:PID:Interface way. 
However talking about "...give the user opportunity to see..." , in my current solution the VID:PID:IFace is the internal "address" with which XCS handles the connection, not the "display name" in the PortPicker dialog, which shows device "name and manufacturer (plus Interface)".
I would be no problem at all to change the display-name to VID:PID:Iface as well, however I made it the other way because this is how UsbSerial devices are displayed in xcs up to now and I wanted first to add the functionality without changing too much else.
I will publish my solution as a pull request. Please review, be thouroug but not too hard on me, I am not experienced in such endeavours yet. ;-)
For all i have tested so far my code works nicely with a single port UsbSerial adapter as well as with the 4-port FTDI4232.

closed 2022-08-19 by MaxKellermann

iltis42, 2022-08-21:
We have tried the latest change with an android build based on the latest changeset (2), but still we see only one port, ttyS1, the console port of this board, but not ttyS3,7,9 as the board in principle had. 
Any idea how to debug this further ? 
Would like to isolate the issue if its an Android issue or related to XCSoar. The Android version is android 11. 
The board is equipped with rockchip RK3566 featuring 10 UARTS and 3 of them are accessible such as UART3,7 and 9 i would like to connect with FLARM or other GPS sources.
Launching the application from 'Android-SerialPort-Api' can see this devices (4):
https://code.google.com/archive/p/android-serialport-api/downloads
https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/android-serialport-api/SerialPort_1.1.apk
What we are missing here to get those devices accessible in XCSoar, or how can we help to debug this further, maybe add some logs somewhere in the code to see what android exactly lists for those ports ?

1) Board with native UART's
[AD-C05-V1-2-Datasheet(1).pdf](https://github.com/XCSoar/XCSoar/files/9389023/AD-C05-V1-2-Datasheet.1.pdf)

2) Debug Version XCSoar, latest pull from yesterday
[image]

3) Devices seen
[image]
[image]
[image]

4)
https://user-images.githubusercontent.com/34288016/185790549-fb3256bb-202b-4d29-9cbb-b0c60595c34f.mp4

henrik1g, 2022-08-21:
> We have tried the latest change with an android build based on the latest changeset (2), but still we see only one port, ttyS1, the console port of a relative new android board, but not ttyS3,7,9 as the board in principle had. Any idea how to debug this further ? Would like to isolate the issue if its an Android issue or related to XCSoar. 

I don't think your problem is related to the code-change done here, which is for adressing multiport usb-to-serial converters connected to a USB-OTG port.
What you are talking about is a android board with native serial ports. Different departement. I have no knowledge about that.

iltis42, 2022-08-21:
Okay i see. At least there is now option to connect an FTDI chip to one of the various USB ports, and operate 4 UART's through this. Thanks for the excellent work!

https://de.aliexpress.com/item/1005001928940997.html?spm=a2g0o.productlist.0.0.4f2e3d30HulMdG&algo_pvid=86afdf09-63ef-4e5c-9421-2ac49dfa459b&algo_exp_id=86afdf09-63ef-4e5c-9421-2ac49dfa459b-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000018135316958%22%7D&pdp_npi=2%40dis%21EUR%2117.81%2113.54%21%21%21%21%21%400b0a050b16610861890803821ee1cc%2112000018135316958%21sea&curPageLogUid=FZKzMr2OUtuo

lyusupov, 2022-08-22:
> We have tried the latest change with an android build based on the latest changeset (2), but still we see only one port, ttyS1, the console port of a relative new android board, but not ttyS3,7,9 as the board in principle had.

@iltis42 , certain /dev/ttyXXXX entries may disappear after this PR: https://github.com/XCSoar/XCSoar/pull/891
Make sure that you are not affected.

iltis42, 2022-08-25:
This change is not affecting me, ttyS* does not carry a digit directly after tty, hence it looks like there is epic issue with permissions, tty's with executable rights are filtered from unknown reason, for details see: https://github.com/XCSoar/XCSoar/issues/1002#issuecomment-1227641155
Do you have any idea why it had been designed like this?