XCSoar/XCSoar #971: Automatic unintentional navigation back to start point

fronzee, 2022-07-19:
XCSoar versions having and not having the problem:
7.23, 7.20 have this issue
-------------------------------------------------

Expected behavior
-----------------
Keep navigating to latest active waypoint unless in sector (speed task/ or armed in AAT). 

Actual behavior
---------------
Since using internal GPS of Xiaomi Redmi note 10 OR via bluetooth with lxnav 8030 - task navigation randomly jumps back to start point. When selecting again point to navigate to in task via next-next... - it goes back after a while (not sure interval, but seems like couple minutes-10 - enough not to see it actually happen in flight, but often enough that would annoy you to see that bearing and distance are wrong later because again navigating to START point.
Start time entry/no entry in task page has no effect on this behaviour.

Had no issue with same phone with LXNAV S10 via bluetooth for more than a year and never before whatsoever.

Steps to reproduce the behavior
-------------------------------
hard to tell... Friends have no similiar issues

Do you have any idea what may have caused this?
-----------------------------------------------
Something to do with start time entry in task page?
losing of GPS signal (not sure my device ever loses GPS signal) without restarting xcsoar has smth to do with this?

Do you have an idea how to solve the issue?
-------------------------------------------
no..

MaxKellermann, 2022-08-24:
You really need to post instructions on how to reproduce. We can't help unless you have such a description. Even more so if you're the only one experiencing this problem. Sorry.

lordfolken, 2022-11-15:
No activity for some time. Closing this issue.

closed 2022-11-15 by lordfolken