XCSoar/XCSoar #973: 7.24 crashes in Android 5.0.1 while in background

Joernle, 2022-07-21:
I use 7.24, because this solves a problem with the flarm radar (7.20) for me (see forum). But in Android 5.0.1 the version 7.24 crashes, as soon as I try to switch to other apps (using the menu button/square near the home button of Android). I don´t have to switch to another app, it even crashes when I look at the overview and then try to get back to XCSoar immediately.

Protection for background/standby activity is set.

Under Android 12 it´s fine.

MaxKellermann, 2022-07-21:
Your bug report is useless without a system log (logcat).

Joernle, 2022-07-21:
Sorry Max, I´m no pro and don´t know where to get this log. I guess you don´t mean xcsoar.log? I searched and now installed Logcat Reader. But this seems to require Root Access or some Software on a PC!? Or do I think too complicated?

Joernle, 2022-07-21:
Now I tried even harder and hope, I have a good result: I was able to run Logcat Reader and did the following:
- Start recording in Logcat Reader
- Start XCSoar
- Finish XCSoar in the normal way
- Start XCSoar
- Go to the overview of running apps with the square near the home button of Android
- (XCSoar crashes)
- Stop recording in Logcat Reader
- Export the file (attached)
I hope this is what you meant..
[logcat_07-21-2022_14-59-12.txt](https://github.com/XCSoar/XCSoar/files/9159584/logcat_07-21-2022_14-59-12.txt)
.

MaxKellermann, 2022-07-21:
This is a bug in the graphics driver of your phone. These drivers are usually proprietary and not open source, which makes it impossible for us to provide support or even further analysis of the problem.
Sorry, this is not something that can be fixed in XCSoar.

closed 2022-07-21 by MaxKellermann

Joernle, 2022-07-21:
Ok, that´s sad. But I don´t understand, why this worked fine for years until 7.20 - so I guess it has at least something to do with changes within XCSoar from 7.20 to 7.24 - even if it´s not possible to fix it.

MaxKellermann, 2022-07-21:
The graphics initalization code has been improved, and the graphics driver is now called in a slightly different way. Obviously in a way that reveals a bug in the graphics driver. But we won't roll back those changes, because they are correct, and the driver is wrong.
It's always sad when you depend on a proprietary piece of software whose vendor never fixes any bugs, or has given up on your product because supporting already-sold products doesn't earn him any money to pay his engineers. That's why it's important to have fully free software stacks. Non-technical people (who havn't lived through this sadness) don't understand the importance of software freedom.

Joernle, 2022-07-21:
Ok, thanks Max for the explanation. I sure understand that you improve XCSoar and this might sometimes cause problems on old devices.

I would like to change to a newer device, but up to now I didn´t find one which is as good as my Huawei MediaPad X2 (in the 7'' display category with bright display and long-lasting battery). So I will have to decide if I can live with the driver-bug (not using any other app parallel) or switch to a new, but smaller tablet.

Thanks anyway for continuing your work on XCSoar!