XCSoar/XCSoar #977: TC loss as additional info to TC gain

hjr, 2022-08-05:
XCSoar version
--------------
7.23  as well as 7.24

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
I'd like to see the over all altitude loss while thermalling and trying to catch a thermal. Mathematically spoken the integral of all vario values below zero freo TC begon to end.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
Comparing and optimizing thermalling strategies.

Writing this in the name of Michael Godoj mgodoj@gmx.net

bomilkar, 2022-08-22:
Is this really useful? Varios have different time constants which then results in different, non-comparable values.
And **how** would this support optimizing thermalling strategies?

MaxKellermann, 2022-08-22:
I was wondering the same. I thought if anything, then the result would be bogus and unusable. What counts is the altitude difference between now and when you entered the thermal. Which we already have, and which @hjr's PR breaks.

lordfolken, 2022-08-23:
I know this feature from seeyou. Basically this shows you how efficentlty you thermal by showing you how many meters you spent circling losing altitude. 
I' don't think he means to aggregate all varios, but just the loss from one vario.

bomilkar, 2022-08-23:
>  I' don't think he means to aggregate all varios, but just the loss from one vario.

A fast vario (short time constant) may show huge losses while a slow vario doesn't show any losses at all.
So what is this saying about my thermaling?

MaxKellermann, 2022-08-23:
> Just one vario: A fast vario (short time constant) may show huge losses while a slow vario doesn't show any losses at all. So what is this saying about my thermaling?

Exactly my point. That's where this idea fails. I don't get it at all.

bomilkar, 2022-08-24:
This is confusing. Your buddy seem to give you inconsistent information on what he has/wants.
You are talking about WeGlide's flight statistics. Do you mean this:
[image]
This is a statistical analysis for each of the 6 legs. Not exactly clear what counts as an "attempt".

The value in the "TC Gain" InfoBox can be that same value at the end of your attempt. And yes, it might be negative.

In any case: how can that change/improve my thermalling? Should I avoid attempts? Should I keep circling if I lose? It's information after the fact!

hjr, 2022-08-24:
It is not about the attempts as far as I understood.

bomilkar, 2022-08-24:
Then what???
----
Liebe Grüße,
Ronald Niederhagen

hjr, 2022-08-24:
I forwarded the question to Michael?

hjr, 2022-08-24:
Hallo Helmut,

interessante Diskussion!
Ja, die Daten sollen von der Höhenmessung kommen, und nicht aus einem Strom von Variometerdaten.

lordfolken hat schon erkannt, daß sich das auf CU Statistikauswertungen bezieht. Hier ein Beispiel eines 700 km Fluges:

[image]

Beim Kreisen habe ich akkumuliert 14841 Höhenmeter gewonnen, aber auch -1520 Höhenmeter verloren. Das sind ca 10%!
Dabei sind die Suchkreise (Tries <45s) von -310 Höhenmeter nicht mitgezählt.

Im Vergleich mit guten Wettbewerbspiloten verliere ich zuviel Höhe beim Kreisen. Um Herauszufinden wann genau in jedem Kreis das passiert, wünsche ich mir diese TC LOSS Echtzeitanzeige.

Ich hoffe, daß das Sinn macht.

VG  Michael

bomilkar, 2022-08-26:
I guess I understand your intend. However I still don't see any value producing this number.
You say you "lost" 1520 m while circling. That is accumulating altitude change below a threshold of 0 m/s. Why not pick a threshold of -0.6 m/s or +0.3 m/s. You see this threshold is entirely arbitrary and so is that "loss" number.
You say minimizing that loss number would make you a better pilot. All that counts is how much time (say 200 sec) you spend to gain altitude (say 500 m). Since thermals are (almost) never round or endless streams the best path may well lead through regions of lower climb rate. Travel speed (hence performance) depends on expected climb rate, and nothing else. That's the core of MacCready's theory. Anyway, better pilots usually spend less time circling.
Imagine 2 gliders in the same thermal: one Ka8 and an eta. They will follow different paths and perhaps the Ka8's loss number is smaller.
So I would not implement this feature since it's bogus at best, perhaps even confusing. The fact that SeeYou (and maybe others) calculate that number doesn't mean anything to me. (I don't jump off a cliff just because somebody else does.)
But in the end: it's not my call.