XCSoar/XCSoar #980: Including Thermal Hotspots to aispaces

Blaubart, 2022-08-07:
Yesterday I flew and it quickly became blue for us. From here (https://www.thermalmap.info) I installed a map in XCSoar on my OpenVario, on which statistically evaluated accumulation points for thermals are entered. The hotspots are loaded as airspace "unknown". That was extremely helpful yesterday, as I found thermals pretty much exactly where they are marked at low altitudes. The problem with this, however, is that these hotspots are displayed regardless of the zoom size. If you zoom the map to more than 150km, so many points are loaded that the processor of the OpenVario is overloaded and the OpemVario no longer reacts to any input. Therefore, I think the idea of ​​introducing a new airspace "Thermal Hotspots" that is only displayed up to a zoom of 60km would be good. That would be absolutely sufficient, because from about 30km the hotspots are no longer clearly recognizable anyway.

groundhogxc, 2022-08-17:
I checked their website, you should be able to download the data as CUP file instead of as an airspace file. With CUP (waypoint) format you should not have this problem.

Blaubart, 2022-08-18:
I know, but as waypoints they're almost undetectable as thermal hotspots. They are only shown as a small cross. In addition, next to each point is the name, e.g. H41. This makes the map confusing.

lordfolken, 2022-08-18:
Yeah well, the airspace renderer was designed to run on IPAQs with 16mb ram. So adding almost 10000 airspaces to it may not have been an envisioned use case. :)

```bash
grep AC thermalmap-17-8-2022_235343-xcsoar.txt  | wc -l
9497
```

lordfolken, 2022-08-18:
Maybe someone can valgrind the thing, and see where we are loosing cpu time?

Blaubart, 2022-08-18:
Thanks for your information. But wouldn't it also make sense not to load the many hotspots with a large zoom in the first place? Then they're not helpful. That would not only save computing power, but might also avoid problems in the future.

Blaubart, 2022-08-29:
ok, i think this is unnecessary now, max found a much better solution

https://github.com/XCSoar/XCSoar/commit/0d305753940bef432b24553380333af1b95f2fd7#commitcomment-82462064

closed 2022-08-29 by Blaubart