XCSoar/XCSoar #981: altitude needed bar

cerio1992, 2022-08-11:
XCSoar version
--------------

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Selection of "type" of "altitude needs bar" on the LH side of the map.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
just to add a selection in the configuration menu to give the user the possibility to chose the setting that they prefer

lordfolken, 2022-08-15:
Can you clarify this please?

lordfolken, 2022-08-16:
So the bar you are shown is the "Final Glide Bar" its purpose is to show the complete altitude needed for final glide. 
There is a setting "Auto" that only shows the final glide bar when in final glide range.
The information you want to know is in the Next Altitude ariival/difference/required infoboxes, which will show you information concerning the next waypoint.

cerio1992, 2022-08-16:
Yeah, I know that this information is displayed in the info boxes… but I think that should be much better to have a “graphic” indication about that info… green/up good, red/down bad… and in this ways I can select another type of information on the info boxes…

MaxKellermann, 2022-08-23:
The next turnpoint of the current task is an arbitrary point on the task just like any other, and the arrival altitude there would be of no particular interest to me. What is interesting is the arrival altitude to all reachable landing sites, and XCSoar already has several graphic and textual representations of that.
I don't like this feature request, but since XCSoar is open source, you're free to take XCSoar's source code and develop the feature for yourself. If you feel like sharing your code with others, come here and submit a PR.

closed 2022-08-23 by MaxKellermann