XCSoar/XCSoar #982: Waypoints replaced by blue dots with smaller zoom values

MikeBravo15, 2022-08-13:
XCSoar versions having and not having the problem
-------------------------------------------------

7.24

Actual/Expected behavior
-----------------

Since intruduction of the 7.24 version of XCS, waypoints got a new symbol, which I find nice. Hovever, when you zoom out, symbols are replaced with big blue dots. I find the dots far too big for my screen - they clutter the view with smaller zoom and also overlap landables. Blue dots are OK, just please make them smaller or size selectable. Thank you

Steps to reproduce the behavior
-------------------------------

Zoom out on map view

lordfolken, 2022-08-15:
The problem is that we lack any sort of scaling for images on screens. So too large for you might be too small for someone else.

MikeBravo15, 2022-08-17:
As at lower zoom values there is just blue circle displayed, is it possible to set it in the same manner as Landable size?

MikeBravo15, 2022-08-23:
Well, I would call it a regression than enhancement as at smaller zoom I usually do not look for waypoints not important for the current task and far away from my current position. I would even kill airspace information boxes at extreme small zoom to enhance clarity. What was the idea behind the blue dots?

MikeBravo15, 2022-08-23:
[image]

MaxKellermann, 2022-08-23:
This... looks awful!

MaxKellermann, 2022-08-23:
I can't reproduce this look. When I zoom out, I don't see those circles.

MikeBravo15, 2022-08-23:
Shall I post both .cup files/other settins? I run XCS on Pixel 4a and Nokia 8 - same behavior

lordfolken, 2022-08-23:
https://github.com/XCSoar/XCSoar/blob/master/build/resource.mk#L42 is the line that resizes all the map icons.
we may need different ones for different sizes. But as long as we don't  have any sort of bitmap/dpi support this is a stop gap solution.

lordfolken, 2022-08-23:
-dpi=400 and i can get something like that on my laptop.

MaxKellermann, 2022-08-24:
I can't reproduce with -dpi=400, and I have no idea where this purple circle comes from.

Scumi, 2022-08-24:
Depends if you have a waypoint file selected with certain waypoint types.

MaxKellermann, 2022-08-24:
Always post files and settings. Not doing so wastes the time of developers for asking questions and waiting for answers, and it delays the bug fix a lot.

lordfolken, 2022-08-24:
These are the waypoints. The previous icon was 3x3 pixel. This has been resized to 13. This happens after a certain zoom level, where the waypoint type specific waypoints get replaced with generic circles.

MikeBravo15, 2022-08-24:
[wpts.zip](https://github.com/XCSoar/XCSoar/files/9418829/wpts.zip)
One cup is containing waypoints, the secont thermal location - even thermal points (another icon) are changed to big circles

MikeBravo15, 2022-08-24:
[image]
[image]

styblope, 2022-09-06:
I can confirm the issue on my setup. I think it may be a problem of a missing viewbox which frames the relative proportions of the dot.
```diff
--- a/Data/icons/map_small.svg
+++ b/Data/icons/map_small.svg
@@ -10,8 +10,9 @@
    xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
    xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
    version="1.1"
-   width="3"
-   height="3"
+   width="13"
+   height="13"
+   viewBox="-5 -5 13 13"
    id="svg2816"
    inkscape:version="0.48.3.1 r9886"
    sodipodi:docname="map_small.svg">
```
BTW, the SVG includes some redundant data such as extra empty layer and Inkscape specific metadata.

MaxKellermann, 2022-09-06:
I have no idea about SVG, and I don't know what your diff really means - but if it solves the problem, please make that a PR.

> SVG includes some redundant data such as extra empty layer and Inkscape specific metadata

Any cleanup patches are also welcome. I'm a big fan of clean code, and if there's something like "clean SVG", I'm all ears.

I don't know if it's a good idea to remove Inkscape specific metadata - is that maybe helpful for future edits of the file? Since we don't ship SVGs, this probably doesn't matter. If we would ship SVGs (one day), it might be enough to strip unnecessary metadata only for the build output?

commit 2144c354 (2022-09-07): Data/icons/map_small.svg: Add viewBox attribute and file cleanup

`viewBox`, a standard SVG attribute, frames the small dot icon so that it
resizes proportionally to the prevailing icons size. Cleanup removes
unneeded Inkscape metadata and redundant objects. Fixes #982
PR #1041 (2022-09-07): Data/icons/map_small.svg: Add viewBox attribute and file cleanup
`viewBox`, a standard SVG attribute, frames the small dot icon so that it resizes proportionally to the prevailing icons size. Cleanup removes unneeded Inkscape metadata and redundant objects. Fixes #982