XCSoar/XCSoar #986: Contour of airspace object does not reflect the AckDay status

bomilkar, 2022-08-16:
XCSoar versions having and not having the problem
-------------------------------------------------
The issue exists in 7.24 .

Expected behavior
-----------------
I would expect the contour to change depending on the AckDay status of the airspace object. The contour style should not change just because you have approached that object once.

Actual behavior
---------------
The contour (or outline) of an airspace object (ED-R, Danger Area, Class C, …) initially has a certain shading. See this example of an PARA Area (Danger Area):
[image]

When you click to this object, you can “Ack Day” this airspace. That removes the shading from the contour. See the same example below:
[image]

Now, when you click to this object again and “Enable” this airspace, the contour keeps the same style: It will not get it’s shading back.

The contour also changes it’s style when you get inside the “Warning time”. After that it keeps it’s changed shading until the next restart of XCSoar.

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

bomilkar, 2022-08-16:
Here is the NMEA file to reproduce the behavior:
[Airspace_bug1.nmea.txt](https://github.com/XCSoar/XCSoar/files/9353889/Airspace_bug1.nmea.txt)

styblope, 2022-08-19:
This fixes it for me.
```diff --git a/src/Engine/Airspace/AirspaceWarning.cpp b/src/Engine/Airspace/AirspaceWarning.cpp
index 19b93173a0..10b4782fdc 100644
--- a/src/Engine/Airspace/AirspaceWarning.cpp
+++ b/src/Engine/Airspace/AirspaceWarning.cpp
@@ -109,8 +109,7 @@ AirspaceWarning::IsAckExpired() const noexcept

   switch (state) {
   case WARNING_CLEAR:
-    /* no warning at all, assume it's still acked */
-    return false;
+    return true;

   case WARNING_TASK:
   case WARNING_FILTER:```

bomilkar, 2022-08-19:
It fixes the issue at least for the 2 scenarios I described.
Great! Thanks!
I'll try it during the next flights and report back if I discover side effects of this fix.

bomilkar, 2022-08-22:
I tried this fix proposed by @styblope : it seems to fix the issue. I tried different scenarios in flight and in replays on the ground. The behavior of the contour shading/padding looks OK to me. And I didn't notice any bad side effects.

However, there is a commit on this code section from @MaxKellermann 
```
 git diff d71d66d0e120d3772680f3d1baae75ff38454598 src/Engine/Airspace/AirspaceWarning.cpp
```
Is there a conflict?

MaxKellermann, 2022-08-22:
D'oh, my commit e60e682ae8d8e0720f33c5bd75d6077800db7549 introduced this regression, and it was the wrong return value. The method has a confusing definition - `IsAckExpired()` means "is this warning **in**valid and can be ignored". This counter-intuitive code design has caused several bugs already..

closed 2022-08-22 by MaxKellermann