XCSoar/XCSoar #991: missing libc++_shared.so on Android

hjr, 2022-08-18:
Hey,
i am trying to build from master branch for android target.
The apk is built without error, however when started on the device the attached error screen appears claiming missing libc++_shared.so

NDK: android-ndk-r25
SDK: build-tools 29.0.3
         platform android-30

make TARGET=ANDROID ANDROID_SDK=/home/user/Android/Sdk ANDROID_NDK=/home/user/Android/android-ndk-r25

Adding LIBCXX=y did not help. (Should be implicit for Android build anyways)

[image]
Any ideas how to fix this?

_Originally posted by @henrik1g in https://github.com/XCSoar/XCSoar/discussions/985_

lordfolken, 2022-08-18:
Can confirm. Also happens with the artefacts downloaded from the pipelines.

lordfolken, 2022-08-18:
Please start bisecting. Not that many comits since v7.24.

ubx, 2022-08-18:
A good opportunity to learn git bisect.

```
4a821d4342c7371467d9d6f16000a1ff95273455 is the first bad commit
commit 4a821d4342c7371467d9d6f16000a1ff95273455
Author: Max Kellermann <max.kellermann@gmail.com>
Date:   Fri Mar 12 22:22:50 2021 +0100

    build/python/libs: update proj to 9.0.1, libgeotiff to 1.7.1

 build/python/build/libs.py             |   32 +-
 build/thirdparty.py                    |    3 +
 lib/libgeotiff/patches/no_csv_data_dir |   24 -
 lib/libgeotiff/patches/series          |    1 -
 lib/proj/patches/no_data               |   18 +
 lib/proj/patches/no_json               | 2002 ++++++++++++++++++++++++++++++++
 lib/proj/patches/pkg-config            |   55 -
 lib/proj/patches/series                |    3 +-
 8 files changed, 2048 insertions(+), 90 deletions(-)
 delete mode 100644 lib/libgeotiff/patches/no_csv_data_dir
 create mode 100644 lib/proj/patches/no_data
 create mode 100644 lib/proj/patches/no_json
 delete mode 100644 lib/proj/patches/pkg-config
```

closed 2022-08-19 by MaxKellermann