XCSoar/XCSoar #992: XCSoarData dir not accessible due to scoped storage policy of Android 11

henrik1g, 2022-08-18:
While testing I came across not being able to access the XCSoarData directory any more after completely uninstalling and reinstalling XCSoar binary on Android.

It seems to be due to the changed storage policy by Android 11.
See here: 
https://developer.android.com/training/data-storage/use-cases

The xcsoar code mentions in LocalPath.cpp
https://github.com/XCSoar/XCSoar/blob/970470f3ab2529e27200a19306d767348a712c34/src/LocalPath.cpp#L249
>         /* the "legacy" external storage directory is writable (either
>            because this is Android 10 or older, or because the
>            "preserveLegacyExternalStorage" is still in effect) - we
>            can use it */

However that only seems to be true if there is still an old installation of XCSoar that is merely updated. For a complete new install or reinstall. XCSoarData is no longer available.

> Note: If you set preserveLegacyExternalStorage to true, the legacy storage model remains in effect only until the user uninstalls your app. If the user installs or reinstalls your app on a device that runs Android 11, then your app cannot opt out the scoped storage model, regardless of the value of preserveLegacyExternalStorage.

Is there anyway around that? Personally I hope there is, since I would hate losing my whole data directory just by accidentally uninstalling XCSoar.
If there is no way around that and scoped storage is the new way eventually, this should be made more visible, to avoid confused users.

henrik1g, 2022-08-18:
A guess that means. There is no way around it.

So to offer some help for people coming across this post copy the contents of your XCSoarData folder (not the folder itself) to the new location:

/Android/data/org.xcsoar/files/

This seems to work for me so far. However be careful since I "think" uninstalling the application will momentarily erase the whole data directory with it. (->Backup)

fronzee, 2022-08-18:
> scoped storage is the new way
>
> It is, because Google forces this anti-feature down everybody's throats.
>
> should be made more visible
>
> Go ahead and make it more visible. I'm tired of wasting time on this sh*t,
> I won't spend any more time on this.

I did downgrade to 7.20 version from 7.23 due to the same issue. It somehow
would be accessable until first restart of the device, and then after I
would need to reinstall xcsoar to access the folder easily. Otherwise some
other file browser is needed and access to very deep folder structure to
put the airspace and wp files to… very annoying :(

henrik1g, 2022-08-18:
I used this App to move the files. The first time it requested some special permission to access the /Android/data/ folder, but that was self explanatory.
https://play.google.com/store/apps/details?id=com.alphainventor.filemanager&hl=de&gl=US

MaxKellermann, 2022-08-18:
> However be careful since I "think" uninstalling the application will momentarily erase the whole data directory with it.

That's true. Google made this stupid user-hostile "feature" because they want to promote apps that use cloud storage, and use local storage only as a cache for the cloud. If a user-friendly app such as XCSoar suddenly loses your data, you'll never use it again and instead switch to some cloud-native sh*t which sells more Google cloud.

The reasons for my Google frustrations are clearly visible, aren't they?

fronzee, 2022-08-18:
If you use original xcsaordata folder for everything - nothing gets lost,
but then after each restart you need to reinstall the xcsoar… :( but why
its ok with 7.20 from playstore and not with 7.23 from .apk?

MaxKellermann, 2022-08-18:
Because that version uses a different targetSdkVersion, but Google rejects uploading new app versions with that older targetSdkVersion. It's all Google's fault, and it's an artificial problem.

lordfolken, 2022-08-18:
Lk8000 has an interesting solution to this problem. They register lk8000 somehow in the ifle manager. So you can easily access the application directory and copy / paste files. In my opinion that's the most elegant solution in this situation:

[image]

iltis42, 2022-08-22:
Just now running in the same issue, just want to make some logs to get light in the tty device issue hence xcsoar.log won't get updated anymore. And we can't even download .igc files from the logger (tried with Flarm F6) somewhere after 7.20, definitively with 7.23 the problem is there, and we cannot create an .nmea logfile, so went back to 7.20 to get my flight uploaded 2 weeks ago.
Reinstallation of xcsoar doesn't help on my phone.

posener, 2022-08-27:
Is there a way that logs will be in `Android/data/org/xcsoar/media` and not in `Android/data/org/xcsoar/files`? The `files` directory is not available in file browser and prevents from uploading flights to OLC on the phone.

brunotl, 2022-08-29:
> Lk8000 has an interesting solution to this problem. They register lk8000 somehow in the ifle manager. So you can easily access the application directory and copy / paste files. In my opinion that's the most elegant solution in this situation:
> 

@lordfolken : if you want to do the same for XCSoar, you just need to get  [AllFilesDocumentsProvider.java](https://github.com/LK8000/LK8000/blob/master/android/src/org/LK8000/AllFilesDocumentsProvider.java) from LK8000 project, change it's pakage name to org.xcsoar and register it in AndroidManifest.xml like that : 

``` xml
 <application...>
    ....
    <provider
        android:name=".AllFilesDocumentsProvider"
        android:authorities="${applicationId}.allfiles"
        android:exported="true"
        android:grantUriPermissions="true"
        android:permission="android.permission.MANAGE_DOCUMENTS">
      <intent-filter>
        <action android:name="android.content.action.DOCUMENTS_PROVIDER" />
      </intent-filter>
    </provider>
  </application>
```

hoping that would help ... ;-)

elgandoz, 2022-09-16:
> Lk8000 has an interesting solution to this problem. They register lk8000 somehow in the ifle manager. So you can easily access the application directory and copy / paste files. In my opinion that's the most elegant solution in this situation:

That's the solution  [XCTrack](https://xctrack.org/Faq.html) also implemented but that didn't solve completely the problem on Android 13: unlike 12, on this version there's no 3rd party File Manager able to access the folder (as far as I am aware) and only the system one can access, but in read only.
That means you can copy files out on `Android/data/org.xcsoar` but not in it.

To workaround the issue, XCTrack devs implemented a filepicker from within the app, which copies the file from any accessible folder to the user, since the app itself has write permission in its own folder.

In this way it's still possible to import custom airspaces, waypoints, task, etc. by using the phone only.

The only other possible way I could find so far is to connect the phone to a computer, enabling the mass storage and copying the files with the host computer file manager.

brunotl, 2022-09-16:
> That's the solution [XCTrack](https://xctrack.org/Faq.html) also implemented but that didn't solve completely the problem on Android 13: unlike 12, on this version there's no 3rd party File Manager able to access the folder (like the one recommended above) and only the system one can access, but in read only. That means you can copy files out of `Android/data/org.xcsoar` but **not in it**.

using Android 13 AVD, i'm able to delete file and to copy files from Download to internal app storage using the system file manager, so the truth is elsewhere...

elgandoz, 2022-09-16:
> using Android 13 AVD, i'm able to delete file and to copy files from Download to internal app storage using the system file manager, so the truth is elsewhere...

Not sure about AVD, but using a real Google Pixel 6 with the latest Android 13 this is what you get: https://photos.app.goo.gl/Qycp3v7hhY1xm24KA

I'm very happy to be shown a working way,  and be enlightened with the "truth".

schmidi, 2022-09-16:
Which version are you using? The actual in the play store does not yet contain the new document provider.

The apk of the current master branch on my Pixel 4a 5G with Android 13 looks like this:

https://user-images.githubusercontent.com/411354/190812946-8c8386c2-7764-4d61-b9d6-054de43e8bea.mp4

elgandoz, 2022-09-16:
> @elgandoz 
> Which version are you using? The actual in the play store does not yet contain the new document provider.
Yes I had the stable version.
I didn't realise that using the document provider shortcut would allow to copy files within it. I was trying to copy the files by going through Internal storage, hitting the permission wall (like I did for the last 6 months!).
I just installed the latest build from GH actions, and works perfectly, this is great, thanks heaps!

MikeBravo15, 2022-09-25:
I am also on Pixel4a, Android 13. If you need to access /storage/emulated/0/Android/data/org.xcsoar/files, use Mixplorer app. But Google is really complicating my life recently.

jwlc76, 2022-12-18:
Just had this problem after upgrading to a Pixel 7 with Android 13. Best work around was to switch to a free file manager app on Google Play Store called "FV File Explorer". It creates shortcuts to specific app data folders.

JoschPaul, 2023-01-14:
I had the same problem on Anroid 17 Google Pixel 7 Pro, both built-in file manager and total commander failed to find or open the directory under Android/Data that XCsoar uses now.
the file manager mentioned above did the job:
https://github.com/XCSoar/XCSoar/issues/992#issuecomment-1219443018
thanks!

JoschPaul, 2023-01-14:
sorry Android 13 not 17 :-)

lordfolken, 2023-02-08:
Would someone be motivated to document the feature in the manual?
Potentially also an FAQ page on the website, would help.

swerthei, 2023-04-04:
I didn't see anybody mention ADB on the thread. You can use ADB to copy files between the Android download directory and the protected data directory. Just FYI.

iltis42, 2023-04-17:
For flight log downloads (IMHO one of the power usecases that need access): Wouldn't be the "Downloads" folder a better option as Downloads directory e.g. for flight logs in Android targets instead of the 'private considered' XCSoarData?  Afaik this folder is not protected by storage policy, and name also tells us this is the place for downloading things.

MikeBravo15, 2023-10-30:
I recently updated my Pixel 6a to Android 14 and here we go again, no access to android/data/ dirs. FV File Explorer can copy out of android/data/..., but not into it. Sigh

lordfolken, 2023-10-30:
The android asop files app on my pixel 6 can copy into and out of the xcsoar folder, via the linked menu entry. https://github.com/XCSoar/XCSoar/issues/992#issuecomment-1249894456

MikeBravo15, 2023-11-01:
You are right, thank you for the tip

lordfolken, 2023-11-01:
If someone would contribute an faq page to the website, that would be awesome

andrekolodochka, 2023-11-02:
I'm moving from Nexus 5s to Pixel 7 (the screen on 6yo Nexus is dying) and I've hit the same problem. I expected to simply upload files to Google Drive and then download them to Pixel and move them to /storage/emulated/0/Android/data as I was doing it on Nexus, but Pixel didn't allow that. I've installed [X-plore File Manager](https://play.google.com/store/apps/details?id=com.lonelycatgames.Xplore&hl=en&gl=US) and after giving it permissions to the folder was able to move the files. Hope this helps.

brunotl, 2023-11-02:
to add shortcut to android stock file manager on pixel 7, you can use this : https://play.google.com/store/apps/details?id=com.marc.files

coljac, 2023-11-17:
I also struggled with this on my Fairphone 4 using Android 13. None of the alternative file managers were of any use. The build-in file manger can be used with the bizarre workaround of opening a new window, going to split screen, and dragging-and-dropping into the `org.xcsoar/files` folder. I say bizarre since they locked down every other avenue I could find, yet this turns out to be relatively easy.

The capability of XCSoar to import files from /Downloads or elsewhere would be appreciated.

andrekolodochka, 2024-03-04:
Update on [my comment above](https://github.com/XCSoar/XCSoar/issues/992#issuecomment-1790519320), if you are still struggling with giving file managers access to your Android/data folder - [this solution](https://www.reddit.com/r/AndroidQuestions/comments/qk9n44/comment/kigffhj/) worked for me:
1. Open X-plore.
2. App manager.
3. System.
4. Search for "files" (it didn't allow me to open ot but it doesn't matter).
5. Long press "files" until a menu appears.
6. App system info.
7. Press the 3 dots and press "uninstall updates".

jkaljundi, 2024-05-24:
Since Android 14 it seems none of the hacks to access XCSoarData and IGC files works anymore. Not the external file managers, Files app updates uninstall or access via USB. Guess there is now no way to access the IGC logs anymore?

iltis42, 2024-05-24:
The way XCSoar might solve this could be using a more public folder like Downloads.

jkaljundi, 2024-05-24:
I don't think Downloads is correct either, as we also need easily accessible directories to write all the Site Files to. These are not downloads. I'm not sure even how and where to write stuff like map database, waypoints, airspaces and so on on Android nowadays?

jkaljundi, 2024-05-24:
> Install this https://play.google.com/store/apps/details?id=com.marc.files
> 
> this launches the original ASOP files app. In the hamburger menu you will find xcsoar, where you can copy and paste files.

Cool, thanks! Indeed this way I can find both the IGC files and the waypoints. Much appreciated. Sorry for the semi-wrong alarm :)

Any way this could be done without the com.marc.files pp in the future?

lordfolken, 2024-05-24:
In the end this thing launches just an intent. Unfortunately its not possible to pre-select a directory... 

I played around with the long press option on the Android App icon. There all the intents could be exposed:
 - Fly
 - Simulator
 -  XCSoarData

The other option would be to make a button in the site-config config panel, that launches this intent on android, and xdg-open on *linux platforms.

lordfolken, 2024-08-08:
A potential solution for the storage accesbility problem in the linked PR, now using Android/media/org.xcsoar directory for storage. See also #1524

lordfolken, 2024-08-13:
This is now merged in 7.43.

closed 2024-08-13 by lordfolken