XCSoar/XCSoar #993: How could the XCM map labels (

dpluigi, 2022-08-18:
XCSoar version
--------------

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

lordfolken, 2022-08-18:
you have to write outside the comments <- ->.

dpluigi, 2022-08-18:
How could the XCM map labels font size (city, ...)?

The waypoint font size can be scaled by the user in Config, the map text is not scalable and is difficult to read even on a 6.8" phablet like the Samsung S22 Ultra.
Can this be implemented in XCSoar or does it have to be done at the XCM map generation stage?

lordfolken, 2022-08-18:
There is some infromation in the manifest. But this mainly concerns the zoom levels of certain layers. 
https://mapgen-data.sigkill.ch/manifest

Font size / scaling is currently a big issue as many elements aren't depending on the DPI of the display.

MaxKellermann, 2022-08-24:
Incomplete bug report. Contains no text atl all.

closed 2022-08-24 by MaxKellermann